OpenBCI/OpenBCI_GUI #607: GUI: Failed to connect to data source

BCIstorm, 2019-09-25:
Hello,
 I'm trying to run the standalone GUI (latest version 4.1.5) on Windows 10 and I can not get the GUI to connect to the Cyton board. Always get the Error message:

> Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN.

It was working on an older version (I believe it was prior to version 4) where one needed to move the `OpenBCIHUB` and `OpenBCI_GUI` folders to the `ProgramFiles` folder and check the `Run as Administrator` box. Is it still necessary? And is it still necessary to first execute the HUB and only then the GUI? 

Here is the Log from 3 consecutive attempts:

```
Console Log Started at Local Time: 2019-09-25_17-05-19
Screen Resolution: 1280 X 1024
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 415, Latest Version: 415
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\Pedro\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM4
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Hub: processConnect: made it -- 10455 code: 200
Hub: setBoardType(): sending 'cyton -- 10456
Hub: processBoardType: success -- 10456
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v1.0.0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
SoftwareSettings: 3 active widgets saved!
Closing any open SD file. Writing 'j' to OpenBCI.
Hub: sendCommand(char): sending 'j'
Hub: processCommand: success -- 11141
Disconnecting serial...
Hub: processCommand: success -- 11193
[ERROR]: Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN.
The file "C:\Users\Pedro\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM1
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Closing any open SD file. Writing 'j' to OpenBCI.
Disconnecting serial...
[ERROR]: Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN.
The file "C:\Users\Pedro\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM4
Settings: LogFileMaxDuration = 60 minutes
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Hub: processConnect: made it -- 22706 code: 200
Hub: setBoardType(): sending 'cyton -- 22706
Hub: processBoardType: success -- 22707
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v1.0.0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
SoftwareSettings: 3 active widgets saved!
Closing any open SD file. Writing 'j' to OpenBCI.
Hub: sendCommand(char): sending 'j'
Hub: processCommand: success -- 23144
Disconnecting serial...
Hub: processCommand: success -- 23196
[ERROR]: Failed to connect to data source. Check that the device is powered on and in range. Also, try pressing AUTOSCAN.
The file "C:\Users\Pedro\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
ConsoleLog: Copying console log to clipboard!

```

Any ideas on what might be causing the GUI not to connect to the Cyton board and do Live streaming?

retiutut, 2019-09-25:
Why didn't you use the new Issue template?

retiutut, 2019-09-25:
Having a hard time knowing where to start and how to help because the information is not organized.

retiutut, 2019-09-25:
Please make a new issue and use the template.

closed 2019-09-25 by retiutut