OpenBCI/OpenBCI_GUI #612: unable link openbci_hub

jh537, 2019-09-29:
> ## Problem
> 
> Dongel not connect, no serial connection
> ## Expected
> 
> In a perfect world, what do you expect to happen.
> ## Operating System and Version
> 
> kubuntu 19.04
> ## GUI Version
> 
> 4.15
> ## Running standalone app
> 
> Are you running the downloaded app or are you running from Processing or Visual Studio Code?
> no
> ## Type of OpenBCI Board
> 
> Cyton
> ## Are you using a WiFi Shield?
> 
> No
> ## Console Log
> 
> Paste any relevant text from the console window here
> Console Log Started at Local Time: 2019-09-29_17-05-26
> Screen Resolution: 3840 X 2160
> Welcome to the Processing-based OpenBCI GUI!
> For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
> Launching application from local data dir
> OpenBCI_GUI: hubStart: OS Detected: Linux
> Local Version: 415, Latest Version: 415
> GUI is up to date!
> OpenBCI_GUI::Setup: Is RX mulitcast: false
> OpenBCI_GUI::Setup: Has RX joined multicast: false
> OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
> ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
> The file "/home/r7598/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
> OpenBCI_GUI::Control Panel: Playback history file not found or other error.
> OpenBCI_GUI::Setup: Setup is complete!
> Instantiating hub object...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [WARN]: Unable to establish link with the OpenBCI Hub, trying again...
> [ERROR]: Unable to find or connect to Hub. LIVE functionality will be disabled.

the directory is already contained in the original file. even if you proceed according to the instructions, the problem still exists.

jh537, 2019-09-29:
the directory is already contained in the original file. even if you proceed according to the instructions, the problem still exists.

jh537, 2019-09-29:
Console Log Started at Local Time: 2019-09-29_20-11-40
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 415, Latest Version: 415
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/r7598/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[WARN]: Unable to establish link with the OpenBCI Hub, trying again...
[ERROR]: Unable to find or connect to Hub. LIVE functionality will be disabled.

jh537, 2019-09-29:
many thanks in advance
:-)

retiutut, 2019-09-29:
Stop spamming! This is not appropriate behavior. You are having problems running the Hub. Please find a way to run the Hub and it will work. There is no problem with the GUI.

closed 2019-09-29 by retiutut