OpenBCI/OpenBCI_GUI #634: BandPower visualization widget does not work in v4.1.7-beta.0

Otemo, 2019-11-07:
## Problem
BandPower visualization widget could not be displayed

## Expected 

BandPower can be choosed from a pull-down menu and visualizaed.

## Operating System and Version

Windows10

## GUI Version

v4.1.7-beta.0
OpenBCIHub v2.1.0

## Running standalone app

running from Processing

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

retiutut, 2019-11-07:
There is no issue. You need to select the Channels from the ChannelSelector. 

To resolve this issue, I can make sure all channels are on by default for Ganglion.

[image]

retiutut, 2019-11-07:
I found the error in the code that was turning on channels by default. It got moved to an external function. This is happening in all modes
[image]

retiutut, 2019-11-07:
Fixed 🛠. Will be available again in 4.1.7-beta.1 ✅

closed 2019-11-07 by retiutut

Otemo, 2019-11-07:
Hi retitutut:
Sorry for taking your time about the issue.
I appreciate your quick support.

retiutut, 2019-11-07:
It's ok @Otemo. It's better to catch things like this quickly, so you did the right thing!