OpenBCI/OpenBCI_GUI #637: Playback mode does not work

Otemo, 2019-11-07:
## Problem

When I try to select and play data file (eg not only my data but "Sample_Data"), the GUI will not work.  By using an older version of the GUI (eg v4.1.3), I can play any file.

## Operating System and Version

Windows10

## GUI Version

OpenBCI_GUI-4.1.7-beta.0

## Running standalone app

Both (downloaded app and Processing)

## Type of OpenBCI Board

files recorded by Ganglion and Cyton

## Are you using a WiFi Shield?

No

## Console Log

I can't find any suspicious logs.

retiutut, 2019-11-08:
Trying now...

retiutut, 2019-11-08:
I'm not having a problem playing back the sample files.

[image]

Try "Clear All Settings".

If this happens again, share a log file please.

Otemo, 2019-11-08:
Thank you @retiutut.
I have cleared all settings, then i tried it again.
Here are my log file

```
Console Log Started at Local Time: 2019-11-08_09-46-11
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 417, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 5
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
[SUCCESS]: All settings have been cleared!
sourceList: click! 
setInterface: curInterface: -1
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
channel count set to 8
[SUCCESS]: You have selected "OpenBCI-sampleData-2-meditation.txt" for playback. 8 channels found.
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
Playback history file not found. making a new one.
Playback history JSON has been made!
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
Index: 0, Size: 0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
CP: Playback History Size = 1
CP: Playback History Size = 1
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2019-11-08:
I'm not sure where `Index: 0, Size: 0` is coming from....

Can you clear the settings again immediately after loading the GUI and tell me if it fails to Initialize System again?

retiutut, 2019-11-08:
Just tested 4.1.7-beta.0 playback mode using "sampleData-2-meditation" on Windows 10 with no error.

Otemo, 2019-11-08:
I think i successfully Initialize system again ...
```
Console Log Started at Local Time: 2019-11-08_10-31-31
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 417, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
ConsoleLog: Copying console log to clipboard!

```

Otemo, 2019-11-08:
GUI will not work...

retiutut, 2019-11-08:
System initialization is everything that happens after 

```
=================================================
||             INITIALIZING SYSTEM             ||
=================================================
```

retiutut, 2019-11-08:
Not sure what to tell you, because I cannot reproduce this error and it works fine on 3 operating systems for me.

Otemo, 2019-11-08:
same phenomenon...
do you want to see log data when GUI(v4.1.3) works well?

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
Index: 0, Size: 0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
CP: Playback History Size = 1
CP: Playback History Size = 1
ConsoleLog: Console closed!

retiutut, 2019-11-08:
@Otemo Do you know how to launch the GUI properly from Processing or Visual Studio Code?

If so, it's time to do some more advanced debugging.

- Clone or Fork/Fetch the latest development branch https://github.com/OpenBCI/OpenBCI_GUI/tree/development
- Go into Debugging.pde and change `boolean isVerbose = false;` to `true`
- Try to load sample data again
- I want to see Verbose system init console log

Otemo, 2019-11-08:
Thank you @retiutut ,
I tried it. I hope I'm doing what makes sense
```
Console Log Started at Local Time: 2019-11-08_11-20-27
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
21.216
35.393997
38.76
42.228
45.696
49.164
52.632
59.772
63.75
66.81
70.278
73.644
77.112
80.478004
83.946
87.312004
90.678
94.145996
97.512
100.979996
104.346
107.712
111.18
114.546
118.014
121.380005
124.746
128.112
131.47801
134.946
138.312
141.678
145.146
148.614
151.878
155.346
158.814
Local Version: 417, Latest Version: 416
GUI is up to date!
162.282
165.64801
169.11601
172.584
176.052
179.622
182.98799
186.456
189.924
193.392
196.86
200.328
203.796
207.26399
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
214.40399
214.71
218.28
221.646
225.114
228.378
231.84601
235.212
238.578
242.04599
245.412
248.88
252.144
255.61201
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
[SUCCESS]: All settings have been cleared!
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
channel count set to 8
[SUCCESS]: You have selected "OpenBCI-sampleData-2-meditation.txt" for playback. 8 channels found.
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
Playback history file not found. making a new one.
Playback history JSON has been made!
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 24318
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
OpenBCI_GUI: initSystem: Initializing core data objects
OpenBCI_GUI: initSystem: -- Init 1 -- 24743
OpenBCI_GUI: initSystem: Initializing FFT data objects
OpenBCI_GUI: initSystem: -- Init 2 -- 24750
OpenBCI_GUI: initSystem: Closing ControlPanel...
OpenBCI_GUI: initSystem: Initializing comms with hub....
OpenBCI_GUI: initSystem: -- Init 3 -- 24750
ChannelController: createChannelSettingButtons: creating channel setting buttons...
comPorts = []
Index: 0, Size: 0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
OpenBCI_GUI: stopRunning: stop running...
CP: Playback History Size = 1
CP: Playback History Size = 1
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2019-11-08:
```
comPorts = []
```

I think maybe the networking widget is crashing because it can't find any comPorts, my Windows 10 machine shows `comPorts = [COM1, COM4]`

Otemo, 2019-11-08:
Thank you for your advice.
Here are log data in older version of the GUI (eg v4.1.3). It works well. 
However, we can find  "comPorts = []".　
```
Console Log Started at Local Time: 2019-11-08_12-24-34
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/OpenBCI%20Software/
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
17.85
35.496002
38.76
42.228
45.594
49.061996
52.427998
59.67
62.934002
66.402
69.869995
73.338005
76.806
80.376
83.742
87.21
90.678
94.145996
97.614006
101.082
104.549995
108.018
111.486
114.954
118.524
121.89
125.358
128.826
132.294
135.76201
139.23
142.596
146.064
149.43001
152.89801
156.264
159.62999
162.99599
166.464
169.83
173.298
176.76599
Local Version: 413, Latest Version: 416
GUI needs to be updated. Download at https://github.com/OpenBCI/OpenBCI_GUI/releases/latest
180.234
183.804
190.23
191.148
194.616
198.084
201.45001
204.918
208.488
211.956
215.32199
218.79
222.258
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 1
OpenBCI_GUI::Setup: Setup is complete!
236.028
236.232
239.7
243.168
246.636
250.10399
253.572
257.142
Instantiating hub object...
Hub: clientEvent: handshake complete
[SUCCESS]: All settings have been cleared!
sourceList: click! 
setInterface: curInterface: -1
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
channel count set to 8
[SUCCESS]: You have selected "OpenBCI-sampleData-2-meditation.txt" for playback. 8 channels found.
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
Playback history file not found. making a new one.
Playback history JSON has been made!
Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 25406
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
OpenBCI_GUI: initSystem: Initializing core data objects
OpenBCI_GUI: initSystem: -- Init 1 -- 25999
OpenBCI_GUI: initSystem: Initializing FFT data objects
OpenBCI_GUI: initSystem: -- Init 2 -- 26007
OpenBCI_GUI: initSystem: Closing ControlPanel...
OpenBCI_GUI: initSystem: Initializing comms with hub....
OpenBCI_GUI: initSystem: -- Init 3 -- 26008
ChannelController: createChannelSettingButtons: creating channel setting buttons...
comPorts = []
WM: Powering 4 widgets up, and remapping.
WM: Activating widget [0]
WM: Activating widget [1]
WM: Activating widget [2]
WM: Activating widget [3]
OpenBCI_GUI: initSystem: -- Init 4 -- 27138
InitSettings: Saving Default Settings to file!
SoftwareSettings: 4 active widgets saved!
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\PlaybackUserSettings.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
InitSettings: C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\PlaybackUserSettings.json not found or other error.
OpenBCI_GUI: initSystem: -- Init 5 -- Load settings error: File not found. 27143
[DEFAULT]: The new data source is PLAYBACK (from file) and NCHAN = [8]. User settings error: Default Settings Loaded.
TopNav: Updated Settings Button Position
OpenBCI_GUI: getPlaybackDataFromTable: End of playback data file.  Starting over...
number of indexes 175954
Finished filling hashmap
^^^GUI update process file has occurred
OpenBCI_GUI: setup: RESIZED
WidgetManager.pde: Remapping widgets to container layout...
headPlot: computeWeightFactor: Ielec 0, solution complete with 2000 iterations. max_dVal = 2.2000075E-5
headPlot: computeWeightFactor: Ielec 1, solution complete with 2000 iterations. max_dVal = 2.2178889E-5
headPlot: computeWeightFactor: Ielec 2, solution complete with 2000 iterations. max_dVal = 3.762841E-5
headPlot: computeWeightFactor: Ielec 3, solution complete with 2000 iterations. max_dVal = 3.7616494E-5
headPlot: computeWeightFactor: Ielec 4, solution complete with 2000 iterations. max_dVal = 1.33931635E-5
headPlot: computeWeightFactor: Ielec 5, solution complete with 2000 iterations. max_dVal = 1.3670326E-5
Copying console log to clipboard!

```

Otemo, 2019-11-08:
I noticed a new fact. If the USB dongle is inserted in the PC, Playback mode will function normally. The following log data is when the playback file is read with BLED112 mounted on the PC.
```
Console Log Started at Local Time: 2019-11-08_18-19-48
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
19.278
33.048
37.026
40.494003
43.86
51.0
54.264
57.732002
61.199997
64.668
68.850006
71.604004
75.072
78.54
82.007996
85.476006
88.944
92.412
95.88
99.348
102.816
106.284
109.854
113.22
116.586
120.054
123.522
126.99
130.356
133.824
137.394
140.76
144.12599
147.59401
151.062
154.53
157.79399
161.262
Local Version: 417, Latest Version: 416
GUI is up to date!
164.73001
168.198
171.666
175.134
178.602
182.06999
185.538
188.904
192.37201
196.452
199.308
202.776
206.24399
209.71199
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 1
OpenBCI_GUI::Setup: Setup is complete!
219.50401
219.708
223.074
226.64401
230.01001
233.478
236.946
240.414
243.88199
247.35
250.81801
254.28601
257.754
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
[SUCCESS]: All settings have been cleared!
sourceList: click! 
setInterface: curInterface: -1
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
channel count set to 8
[SUCCESS]: You have selected "OpenBCI-sampleData-2-meditation.txt" for playback. 8 channels found.
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
Playback history file not found. making a new one.
Playback history JSON has been made!
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: initSystem: -- Init 0 -- 19350
OpenBCI_GUI: initSystem: Preparing data variables...
OpenBCI_GUI: initSystem: loading playback data from C:\Users\koiwa\Documents\OpenBCI_GUI\Sample_Data\OpenBCI-sampleData-2-meditation.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 8
Sample rate set to 250
readCSV: %Sample Rate = 250.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  175953 rows of data, which is 704 seconds of EEG data
OpenBCI_GUI: initSystem: Initializing core data objects
OpenBCI_GUI: initSystem: -- Init 1 -- 20444
OpenBCI_GUI: initSystem: Initializing FFT data objects
OpenBCI_GUI: initSystem: -- Init 2 -- 20453
OpenBCI_GUI: initSystem: Closing ControlPanel...
OpenBCI_GUI: initSystem: Initializing comms with hub....
OpenBCI_GUI: initSystem: -- Init 3 -- 20454
ChannelController: createChannelSettingButtons: creating channel setting buttons...
comPorts = [COM3]
WM: Powering 4 widgets up, and remapping.
WM: Activating widget [0]
WM: Activating widget [1]
WM: Activating widget [2]
WM: Activating widget [3]
OpenBCI_GUI: initSystem: -- Init 4 -- 21846
InitSettings: Saving Default Settings to file!
SoftwareSettings: 4 active widgets saved!
The file "C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\PlaybackUserSettings.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
InitSettings: C:\Users\koiwa\Documents\OpenBCI_GUI\Settings\PlaybackUserSettings.json not found or other error.
OpenBCI_GUI: initSystem: -- Init 5 -- Load settings error: File not found. 21856
[DEFAULT]: The new data source is PLAYBACK (from file) and NCHAN = [8]. User Settings Error: Default Settings Loaded.
TopNav: Updated Settings Button Position
OpenBCI_GUI: getPlaybackDataFromTable: End of playback data file.  Starting over...
number of indexes 175954
Finished filling hashmap
^^^GUI update process file has occurred
ConsoleLog: Copying console log to clipboard!

```

Otemo, 2019-11-08:
When the OpenBCI USB dongle (for Cyton board) was inserted into the PC, it was confirmed that the playback mode works without problems.
As @retiutut pointed out, there seems to be a problem with comPorts.

retiutut, 2019-11-12:
Replicated! 🔥 🔥 🔥

retiutut, 2019-11-12:
The above fix should do it! Fix will be available in 4.1.7-beta.1.

If there are no comPorts, initialize the serial mode port dropdown with just "None" as the single option.

closed 2019-11-12 by retiutut

Otemo, 2019-11-13:
Thank you @retiutut.
I look forward to 4.1.7-beta.1!

retiutut, 2019-11-13:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.7-beta.2

Otemo, 2019-11-13:
Hi @retiutut :
I have confirmed v4.1.7-beta.2 works well! 
Thank you for your support.