OpenBCI/OpenBCI_GUI #638: ArrayOutOfBoundsException when streaming 16 channels over network widget

Joe-Westra, 2019-11-08:
## Problem

When trying to stream data out over the network using the network widget an ArrayOutOfBoundsException is thrown if all of the following conditions hold:
-The number of channels is 16 (either from playback file or from live data)
-The protocol in the network widget is "UDP", "OSC", or "Serial"
-The Data Type is "Time Series"
-"Filters" are turned on

## Expected 

As the unfiltered data stream works fine if "Filters" are set to the off position, I would expect that filtered data would stream out once the Filters are set to On.

## Operating System and Version

Linux Ubuntu 18.04.3

## GUI Version

4.1.6

## Running standalone app

Running from Processing

## Type of OpenBCI Board

Cyton+Daisy
The same issue is experienced when running via the "playback" function with any recorded data.
NOTE: a 16 channel sample included in the /SampleData would be a nice addition!

## Are you using a WiFi Shield?

Yes

## Console Log

I modified the `sendTimeSeriesData() ` method within DataProcessing to print out some extra information to aid in debugging purposes:
```
// TIME SERIES FILTERED
        } else if (filter==1) {
            if (this.protocol.equals("OSC")) {
                for (int i=0;i<nPointsPerUpdate;i++) {
                    msg.clearArguments();
                    for (int j=0;j<numChan;j++) {
                        msg.add(dataBuffY_filtY_uV[j][start+i]);
                    }
                    try {
                        this.osc.send(msg,this.netaddress);
                    } catch (Exception e) {
                        println(e.getMessage());
                    }
                }
            } else if (this.protocol.equals("UDP")) {
                //Added: provide the length of the array
                println("*length of the array: " + dataBuffY_filtY_uV[0].length);

                for (int i=0;i<nPointsPerUpdate;i++) {
                    if (packetNum == MAXPNUM){
                      packetNum = 0;
                    } else {
                      packetNum += 1; 
                    }
                    //Added: packetNumber included in each sample sent
                    String outputter = "{\"type\":\"eeg\",\"packetNumber\":\""+ packetNum +"\",\"data\":[";

                    for (int j = 0; j < numChan; j++) {
                      //Added: Print the current buffer contents
                      println("*buffer data: " + str(dataBuffY_filtY_uV[j][start+i]) + "\tj: " + j + "\tstart+i: " + str(int(start) + int(i)) );
                        outputter += str(dataBuffY_filtY_uV[j][start+i]);
                        if (j != numChan - 1) {
[... 370 lines cut ...]
*buffer data: -173744.16	j: 10	start+i: 22998
*buffer data: -116275.54	j: 11	start+i: 22998
*buffer data: -156291.53	j: 12	start+i: 22998
*buffer data: -50247.637	j: 13	start+i: 22998
*buffer data: -150550.12	j: 14	start+i: 22998
*buffer data: -133065.23	j: 15	start+i: 22998
*buffer data: -24.709085	j: 0	start+i: 22999
*buffer data: -3.5687938	j: 1	start+i: 22999
*buffer data: -18.075706	j: 2	start+i: 22999
*buffer data: -34.69309	j: 3	start+i: 22999
*buffer data: -18.063953	j: 4	start+i: 22999
*buffer data: -3619.7659	j: 5	start+i: 22999
*buffer data: -619.56305	j: 6	start+i: 22999
*buffer data: -43.389004	j: 7	start+i: 22999
*buffer data: -23.336735	j: 8	start+i: 22999
*buffer data: -49.829514	j: 9	start+i: 22999
*buffer data: -173712.16	j: 10	start+i: 22999
*buffer data: -116236.266	j: 11	start+i: 22999
*buffer data: -156261.28	j: 12	start+i: 22999
*buffer data: -50244.55	j: 13	start+i: 22999
*buffer data: -150537.88	j: 14	start+i: 22999
*buffer data: -133053.81	j: 15	start+i: 22999
ArrayIndexOutOfBoundsException: 23000

```

Joe-Westra, 2019-11-08:
As there is no issue when streaming out either filtered or unfiltered data for 8 channels, it _appears to me_ that the issue is due to the `start` and `end` values being more or less hard-coded at the top of the W_Networking:

```
// Data buffers
    int start = dataBuffY_filtY_uV[0].length-11;
    int end = dataBuffY_filtY_uV[0].length-1;
    int bufferLen = end-start;
    float[] dataToSend = new float[numChan*bufferLen];
```
Any feedback on the relevance of `11` and `1` here?

Joe-Westra, 2019-11-08:
It appears to me that the `start` variable should be set to `dataBuffY_filtY_uV[0].length - nPointsPerUpdate`.  Doing so within Stream.updateNumChan allows this to be updated at the correct point:

```
    private void updateNumChan(int _numChan) {
        numChan = _numChan;
        println("Stream update numChan to " + numChan);
        dataToSend = new float[numChan * nPointsPerUpdate];
        println("nPointsPerUpdate " + nPointsPerUpdate);
        println("dataToSend len: " + numChan * nPointsPerUpdate);

        //Add the code below
        start = dataBuffY_filtY_uV[0].length - nPointsPerUpdate;
    }
```

retiutut, 2019-11-08:
@Joe-Westra Almost thought the above merge was done on the upstream branch for a second. I can test this out today, but I've also gone ahead and made a PR from your development branch.

retiutut, 2019-11-08:
Was unable replicate the ArrayOutofBoundsException on MacOS using OSC or UDP with filter On.

retiutut, 2019-11-08:
@Joe-Westra Unable to replicate this issue on Ubuntu 18.04.3. :/

Joe-Westra, 2019-11-08:
It looks like it might be due to a high sampling rate (1000hz), which is used to calculate nPointsPerUpdate in OpenBCi_GUI.initCoreDataObjects().  What sample rate were you using @retiutut?

retiutut, 2019-11-08:
Cyton+Daisy over Dongle is 125Hz.

retiutut, 2019-11-08:
@Joe-Westra Replicated! 🔥 🔥 🔥

retiutut, 2019-11-08:
Fix confirmed using Streaming via OSC to Max/MSP 16ch @ 1000Hz. 🛠

closed 2019-11-09 by retiutut