OpenBCI/OpenBCI_GUI #639: Problem finding saved recordings on linux

aggo, 2019-11-08:
I am using v4.1.6 of the GUI and I am trying to find the saved recordings on my KDE neon 5.15 system. However, I cannot see them in the Documents/OpenBCI_GUI folder, nor somewhere in the app folder. However, the GUI says there is a file being saved (see screenshot). Any clues would be highly appreciated!
[image]

retiutut, 2019-11-08:
I can assure you the file will be there once you are done recording.

closed 2019-11-08 by retiutut

aggo, 2019-11-08:
done recording = stop streaming, no?

retiutut, 2019-11-08:
In the screenshot above, the data is still being saved to file. Every time the data stream stops, the file is saved. You can't close a file that's still open.

aggo, 2019-11-08:
Yes but there were no files even after closing the GUI altogether..

retiutut, 2019-11-08:
For starters, you did not use the Issue template. Very hard to help when you don't provide required information.

retiutut, 2019-11-08:
https://github.com/OpenBCI/OpenBCI_GUI#system-requirements

retiutut, 2019-11-08:
There are actual issues here, whereas this seems like a wild goose chase.

aggo, 2019-11-08:
Sorry for not using the template. I assumed maybe it's something very obvious that I'm doing wrong. Still assuming this, the place where I should find the recordings is `/home/<user>/Documents/OpenBCI_GUI/Recordings/`, no? Also, once closing the GUI the recordings should be there, right? Also, is the OpenBCI_GUI/Recordings directory structure automatically created? In my case I tried even creating the structure by myself but to no avail. 
This is my console output:
```
sudo ./OpenBCI_GUI 
Console Log Started at Local Time: 2019-11-08_20-36-03
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 416, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/root/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
ControlPanel: Attempting to connect to /dev/ttyUSB0
Settings: LogFileMaxDuration = 60 minutes
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
Hub: processConnect: made it -- 11903 code: 200
Hub: setBoardType(): sending 'cyton -- 11903
Hub: processBoardType: success -- 11904
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v3.0.0
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
LoadGUISettings: Applied Widget 0 to Container 4
LoadGUISettings: Applied Widget 1 to Container 3
LoadGUISettings: Applied Widget 2 to Container 9
Hub: sendCommand(String): sending '/0'
Cyton: setBoardMode to :DEFAULT
LoadGUISettings: Expert Mode Off
Hub: processCommand: success -- 12533
Networking: Protocol mode set to Serial
Networking Data Types || show num dropdowns = 13
Apply Serial Networking Mode
Ch 0, Active, x24, Normal, Include, On, Off
Ch 1, Active, x24, Normal, Include, On, Off
Ch 2, Active, x24, Normal, Include, On, Off
Ch 3, Active, x24, Normal, Include, On, Off
Ch 4, Active, x24, Normal, Include, On, Off
Ch 5, Active, x24, Normal, Include, On, Off
Ch 6, Active, x24, Normal, Include, On, Off
Ch 7, Active, x24, Normal, Include, On, Off
Success writing channel 0
Return code: 200
Success writing channel 1
Return code: 200
Success writing channel 2
Return code: 200
Success writing channel 3
Return code: 200
Success writing channel 4
Return code: 200
Success writing channel 5
Return code: 200
Success writing channel 6
Return code: 200
Success writing channel 7
[DEFAULT]: Data stream started.
Cyton: startDataTransfer(): writing 'b' to the serial port...
Hub: sendCommand(char): sending 'b'
OpenBCI_GUI: openNewLogFile: opened ODF output file: /root/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-08_20-36-05/OpenBCI-RAW-2019-11-08_20-36-20.txt
Hub: processCommand: success -- 18167
[DEFAULT]: Data stream stopped.
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
Hub: processCommand: success -- 23946
-- UDP session started at Fri Nov 08 20:36:05 EET 2019 --
-- bound socket to host:127.0.0.1, port: 51000 --
[19-11-08 20:36:34.115 +0200] close socket < port:51000, address:127.0.0.1 >

Client SocketException: Socket closed
SHUTDOWN HOOK
SoftwareSettings: 3 active widgets saved!
Killed:  3254 pts/2    00:00:01 OpenBCIHub
Killed:  3256 pts/2    00:00:00 OpenBCIHub
Killed:  3288 pts/2    00:00:00 OpenBCIHub
SHUTDOWN HUB
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f7148001270, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f7148012750, refCount 1, unCloseable false]
```

retiutut, 2019-11-08:
I'm seeing the GUI making the file!

/root/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-08_20-36-05/OpenBCI-RAW-2019-11-08_20-36-20.txt

It's right there!

aggo, 2019-11-08:
[image]
But there are no files in the end, maybe I'm looking in the wrong place?

aggo, 2019-11-08:
(I created an issue by the template here, it has some extra info https://github.com/OpenBCI/OpenBCI_GUI/issues/641)