OpenBCI/OpenBCI_GUI #641: Cannot see recording files on linux

aggo, 2019-11-08:
## Problem

Cannot see the recording files under /Documents/OpenBCI_GUI/Recordings.

## Expected 

A new recording file is there after closing the GUI.

## Operating System and Version

Linux
Operating System: KDE neon 5.16
KDE Plasma Version: 5.16.3
KDE Frameworks Version: 5.60.0
Qt Version: 5.12.3
Kernel Version: 4.15.0-55-generic
OS Type: 64-bit
Processors: 8 × Intel® Core™ i7-2670QM CPU @ 2.20GHz
Memory: 15,6 GiB of RAM

## GUI Version

416

## Running standalone app

yes

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

```
sudo ./OpenBCI_GUI 
Console Log Started at Local Time: 2019-11-08_20-36-03
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 416, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/root/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
ControlPanel: Attempting to connect to /dev/ttyUSB0
Settings: LogFileMaxDuration = 60 minutes
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
Hub: processConnect: made it -- 11903 code: 200
Hub: setBoardType(): sending 'cyton -- 11903
Hub: processBoardType: success -- 11904
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v3.0.0
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
LoadGUISettings: Applied Widget 0 to Container 4
LoadGUISettings: Applied Widget 1 to Container 3
LoadGUISettings: Applied Widget 2 to Container 9
Hub: sendCommand(String): sending '/0'
Cyton: setBoardMode to :DEFAULT
LoadGUISettings: Expert Mode Off
Hub: processCommand: success -- 12533
Networking: Protocol mode set to Serial
Networking Data Types || show num dropdowns = 13
Apply Serial Networking Mode
Ch 0, Active, x24, Normal, Include, On, Off
Ch 1, Active, x24, Normal, Include, On, Off
Ch 2, Active, x24, Normal, Include, On, Off
Ch 3, Active, x24, Normal, Include, On, Off
Ch 4, Active, x24, Normal, Include, On, Off
Ch 5, Active, x24, Normal, Include, On, Off
Ch 6, Active, x24, Normal, Include, On, Off
Ch 7, Active, x24, Normal, Include, On, Off
Success writing channel 0
Return code: 200
Success writing channel 1
Return code: 200
Success writing channel 2
Return code: 200
Success writing channel 3
Return code: 200
Success writing channel 4
Return code: 200
Success writing channel 5
Return code: 200
Success writing channel 6
Return code: 200
Success writing channel 7
[DEFAULT]: Data stream started.
Cyton: startDataTransfer(): writing 'b' to the serial port...
Hub: sendCommand(char): sending 'b'
OpenBCI_GUI: openNewLogFile: opened ODF output file: /root/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-08_20-36-05/OpenBCI-RAW-2019-11-08_20-36-20.txt
Hub: processCommand: success -- 18167
[DEFAULT]: Data stream stopped.
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
Hub: processCommand: success -- 23946
-- UDP session started at Fri Nov 08 20:36:05 EET 2019 --
-- bound socket to host:127.0.0.1, port: 51000 --
[19-11-08 20:36:34.115 +0200] close socket < port:51000, address:127.0.0.1 >

Client SocketException: Socket closed
SHUTDOWN HOOK
SoftwareSettings: 3 active widgets saved!
Killed:  3254 pts/2    00:00:01 OpenBCIHub
Killed:  3256 pts/2    00:00:00 OpenBCIHub
Killed:  3288 pts/2    00:00:00 OpenBCIHub
SHUTDOWN HUB
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f7148001270, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f7148012750, refCount 1, unCloseable false]
```

retiutut, 2019-11-08:
[image]

> But there are no files in the end, maybe I'm looking in the wrong place?

Thanks for this screenshot.

retiutut, 2019-11-08:
```
sudo ./OpenBCI_GUI 
```
This appears to happen even when running as admin.

aggo, 2019-11-08:
I will try using some older versions of the GUI, will report back.

aggo, 2019-11-08:
Up to 4.1.2 inclusively it works, from 4.1.3 on not anymore. I suspect it has to do with creating the files locally in the same folder vs creating them in /Documents.

retiutut, 2019-11-08:
I can look into this more, thanks for making a new issue.

retiutut, 2019-11-09:
Could this be caused by directory permissions of some kind? I think the file channel gets opened directly to the `Documents/`directory in 4.1.6. 

This is indicated in the line 
```
OpenBCI_GUI: openNewLogFile: opened ODF output file: /root/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-08_20-36-05/OpenBCI-RAW-2019-11-08_20-36-20.txt
```

aggo, 2019-11-09:
I just managed to make it work by running it without sudo. The output folder changed to the correct one, `/home/<user>/Documents` instead of `/root/Documents`.
```
billy@billy:~/Soft/OpenBCI_GUI$ ./OpenBCI_GUI 
Console Log Started at Local Time: 2019-11-09_10-44-48
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Killed:  6161 ?        00:00:01 OpenBCIHub
Killed:  6163 ?        00:00:00 OpenBCIHub
Killed:  6200 ?        00:00:00 OpenBCIHub
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 416, Latest Version: 416
GUI is up to date!
opening socket failed!
        > address:127.0.0.1, port:51000 [group:null]
        > Address already in use (Bind failed)
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Copying sample data to Documents/OpenBCI_GUI/Sample_Data
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/billy/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
ControlPanel: Attempting to connect to /dev/ttyUSB0
Settings: LogFileMaxDuration = 60 minutes
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
Hub: processConnect: made it -- 14628 code: 200
Hub: setBoardType(): sending 'cyton -- 14628
Hub: processBoardType: success -- 14629
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v3.0.0
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
The file "/home/billy/Documents/OpenBCI_GUI/Settings/CytonUserSettings.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
InitSettings: /home/billy/Documents/OpenBCI_GUI/Settings/CytonUserSettings.json not found or other error.
[DEFAULT]: The new data source is LIVE (from Cyton  v3.0.0) and NCHAN = [8]. User Settings Error: Default Settings Loaded.
[DEFAULT]: Data stream started.
Cyton: startDataTransfer(): writing 'b' to the serial port...
Hub: sendCommand(char): sending 'b'
OpenBCI_GUI: openNewLogFile: opened ODF output file: /home/billy/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-09_10-44-52/OpenBCI-RAW-2019-11-09_10-45-06.txt
Hub: processCommand: success -- 19679
[DEFAULT]: Data stream stopped.
Cyton: startDataTransfer(): writing 's' to the serial port...
Hub: sendCommand(char): sending 's'
Hub: processCommand: success -- 23292
```
I remember having tried this before but then I wasn't able to actually start streaming unless I ran it with sudo.

aggo, 2019-11-09:
Managed to reproduce the other problem: now i cannot start streaming at all unless I use sudo
```
billy@billy:~/Soft/OpenBCI_GUI$ ./OpenBCI_GUI      
Console Log Started at Local Time: 2019-11-09_11-30-09
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
Local Version: 416, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/billy/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to tcp
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
ControlPanel: Attempting to connect to /dev/ttyUSB0
Settings: LogFileMaxDuration = 60 minutes
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Hub: processConnect: made it -- 7866 code: 402
Error in processConnect: RESP_ERROR_UNABLE_TO_CONNECT
Hub: killAndShowMsg: Unable to Connect: Error: Permission denied, cannot open /dev/ttyUSB0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
Closing any open SD file. Writing 'j' to OpenBCI.
Disconnecting serial...
Failed to connect to data source... 1
[ERROR]: Failed to connect to data source fail point 1
[ERROR]: Check that the device is powered on and in range. Also, try AUTOSCAN. Otherwise, Cyton firmware is out of date.
The file "/home/billy/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
```

retiutut, 2019-11-13:
@aggo All good then?

aggo, 2019-11-13:
@retiutut Yes, thank you.

closed 2019-11-13 by retiutut

ihgumilar, 2020-12-15:
Hi ,

I have the following issue. Could anyone help me out ? I also cannot connect. All ports seem to be busy.
Don't know what the issue is.
I tried to run openBCI in another PC with the latest openBCI GUI, it works but not in my new PCs.

Thanks
IG
=======

```
Console Log Started at Local Time: 2020-12-15_18-48-25
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
TopNav: Internet Connection Successful
Local Version: v5.0.1, Latest Version: v5.0.1
GUI is up to date!
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\igum002\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
null
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: COM3
Channel count set to 16
ControlPanel: Attempting to Auto-Connect to Cyton
ControlPanel: Found Cyton Dongle on COM port: COM3
Cyton AutoConnect Button: get_channel
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
[ERROR]: Radios_Config: Serial Port in use. Try another port or unplug/plug dongle.
Failed to connect using COM3
[WARN]: Found a Cyton dongle, but could not connect to the board.
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM3
ControlPanel: Found Cyton Dongle on COM port: COM3
Radios_Config: scan_channels
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
[ERROR]: Radios_Config: Serial Port in use. Try another port or unplug/plug dongle.
Failed to connect using COM3
ConsoleLog: Copying console log to clipboard!

```