OpenBCI/OpenBCI_GUI #642: Synthetic data source does not work

daniel-kukiela, 2019-11-11:
## Problem

Synthetic (algorithmic) data source does not work. Shows an empty window exactly like for issue #637 :
[image]

## Expected 

Ability to use synthetically generated data for testing purposes

## Operating System and Version

Windows 10. As Far as I know it works under Linux (tested by Sentdex). I don't know about MacOS.

## GUI Version

I tried that with 4.1.6, 4.1.5 and development branch + Processing.

## Running standalone app

Downloaded app as well as development branch + Processing.

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

## Console Log

```
Console Log Started at Local Time: 2019-11-11_04-55-37
Screen Resolution: 1920 X 1200
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
Local Version: 416, Latest Version: 416
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 2
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
channel count set to 16
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Index: 0, Size: 0
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
CP: Playback History Size = 2
ConsoleLog: Copying console log to clipboard!

```

This issue was originally reported by Harrison (Sentdex) directly. Filling an issue asked by him to do that.

retiutut, 2019-11-11:
I believe this is caused by the same cause of this issue https://github.com/OpenBCI/OpenBCI_GUI/issues/637

I think it might all be connected to CP5 widget error due to null array in Networking widget.

Temporarily fixed by plugging in at least one serial port (ex. Cyton Dongle).

retiutut, 2019-11-12:
Replicated! 🔥 🔥 🔥

retiutut, 2019-11-12:
It looks like we do have to give the Serial Port dropdown at least 1 item in order to be instantiated, and really this list only gets populated on each session start.

So, let's just add "None" to the list if the GUI was unable to detect any serial ports, which typically happens on Windows where comPorts are named COM1 or COM4. On Mac OS and Linux, there are extraneous comPorts that show up.

```
 void createPortDropdown(String name, List<String> _items, boolean isEmpty) {
        if (isEmpty) _items.add("None"); // Fix #642 and #637
        cp5_networking_portName.addScrollableList(name)
            .setOpen(false)
            .setColorBackground(color(31,69,110)) // text field bg color
            .setColorValueLabel(color(255))       // text color
            .setColorCaptionLabel(color(255))
            .setColorForeground(color(125))    // border color when not selected
            .setColorActive(color(150, 170, 200))       // border color when selected
            // .setColorCursor(color(26,26,26))
            .setSize(itemWidth,(_items.size()+1)*(navH-4))// + maxFreqList.size())
            .setBarHeight(navH-4) //height of top/primary bar
            .setItemHeight(navH-4) //height of all item/dropdown bars
            .addItems(_items) // used to be .addItems(maxFreqList)
            .setVisible(false)
            ;
        cp5_networking_portName.getController(name)
            .getCaptionLabel() //the caption label is the text object in the primary bar
            .toUpperCase(false) //DO NOT AUTOSET TO UPPERCASE!!!
            .setText("None")
            .setFont(h4)
            .setSize(14)
            .getStyle() //need to grab style before affecting the paddingTop
            .setPaddingTop(4)
            ;
        cp5_networking_portName.getController(name)
            .getValueLabel() //the value label is connected to the text objects in the dropdown item bars
            .toUpperCase(false) //DO NOT AUTOSET TO UPPERCASE!!!
            .setText("None")
            .setFont(h5)
            .setSize(12) //set the font size of the item bars to 14pt
            .getStyle() //need to grab style before affecting the paddingTop
            .setPaddingTop(3) //4-pixel vertical offset to center text
            ;
    }
```

retiutut, 2019-11-12:
The above fix has been tested on all 3 major OS.

retiutut, 2019-11-12:
Just closed #637 👍 

Let me get GUI 4.1.7-beta.1 ready for you to try @daniel-kukiela

retiutut, 2019-11-13:
https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v4.1.7-beta.2

closed 2019-11-13 by retiutut