OpenBCI/OpenBCI_GUI #648: Feature: Cyton Impedance Check Headplot

retiutut, 2019-11-14:
**Is your feature request related to a problem? Please describe.**
Takes too long to check impedance on all channels when using Cyton and Cyton+Daisy.

**Describe the solution you'd like**
A way to test impedance on all Cyton channels at once, and possibly display this on a headplot diagram.

**Describe alternatives you've considered**
Current solution requires users to click all 16buttons on and off. Could also implement an Expert Mode keyboard shortcut.

**Additional context**
https://openbci.com/forum/index.php?p=/discussion/2364/feature-request-impedance-check-map#latest

[image]

markopetrovic-neurotech, 2021-06-08:
Cant this be done simply by swapping the eeg data used in Headplot with the global impedence data variable?

  headPlot.setIntensityData_byRef(dataProcessing.data_std_uV, is_railed);
->         headPlot.setIntensityData_byRef(data_elec_imp_ohm, is_railed);

retiutut, 2021-06-08:
@markopetrovic-neurotech It's technically correct but not so simple. If you look at the image above, the issue arises when users choose something other than the default electrode positions. We need to expand the electrode map to something like what's shown above and make it configurable/assignable to channels.

markopetrovic-neurotech, 2021-06-08:
Thanks for the timely responses!

I thought we could load other electrode positions from electrode_position.txt files, or is the issue for mappings with more than 16 electrodes? 

Also I see the position electrode_xy array is being updated when the electrodes are dragged, it seems to be working but I'm not sure if I can tell the difference.

Thanks again, I'm excited for the upcoming milestone and would love to help where I can.

retiutut, 2021-06-08:
> Thanks for the timely responses!
>
> I thought we could load other electrode positions from electrode_position.txt files, or is the issue for mappings with more than 16 electrodes?
>
> Also I see the [...]

Woah... I just learned something new about the GUI! But I think that moving them to an arbitrary XY position is a "hacky" solution in comparison to showing a full 10-20 diagram and clicking the ones we want.

Very interesting.... thanks for sharing again!

markopetrovic-neurotech, 2021-06-08:
Haha happy to help! I see a few developers have moved on from the project then? 

I agree it seems a bit hacky. I wonder if a drop down to select between different text files would be viable.

retiutut, 2021-06-08:
> Haha happy to help! Have a few developers have moved on from the project?
> 
> I agree it seems a bit hacky. I wonder if a drop down to select between different text files would be viable.

https://github.com/OpenBCI/OpenBCI_GUI/issues/966

retiutut, 2025-04-25:
Cyton Impedance widget now exists. This should be done now that the HeadPlot widget has been removed.