OpenBCI/OpenBCI_GUI #649: Can't Replay on Ubuntu

Sentdex, 2019-11-14:
## Problem

Replay data not working on Ubuntu

## Expected 

Replay data is expected to work :D

## Operating System and Version

Ubuntu 18.04

## GUI Version

v4.1.7-beta.2

## Running standalone app

standalone

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

./OpenBCI_GUI 
Console Log Started at Local Time: 2019-11-13_21-05-08
Screen Resolution: 2880 X 1620
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
TopNav: Internet Connection Successful
Local Version: 417, Latest Version: 417
GUI is up to date!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected /home/h/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-13_20-48-49/OpenBCI-RAW-2019-11-13_20-50-13.txt
OpenBCI_GUI: initSystem: loading playback data from /home/h/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2019-11-13_20-48-49/OpenBCI-RAW-2019-11-13_20-50-13.txt
readCSV: %OpenBCI Raw EEG Data
readCSV: %Number of channels = 16
Sample rate set to 125
readCSV: %Sample Rate = 125.0 Hz
readCSV: %First Column = SampleIndex
readCSV: %Last Column = Timestamp 
readCSV: %Other Columns = EEG data in microvolts followed by Accel Data (in G) interleaved with Aux Data
OpenBCI_GUI: initSystem: loading complete.  82822 rows of data, which is 663 seconds of EEG data
channel count set to 16
[SUCCESS]: You have selected "OpenBCI-RAW-2019-11-13_20-50-13.txt" for playback. 16 channels found.
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.selectCallback(PApplet.java:6712)
	at processing.core.PApplet.access$1(PApplet.java:6705)
	at processing.core.PApplet$4.run(PApplet.java:6616)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.RuntimeException: A JSONObject text must begin with '{'
	at processing.data.JSONObject.<init>(JSONObject.java:242)
	at processing.data.JSONObject.<init>(JSONObject.java:226)
	at processing.core.PApplet.loadJSONObject(PApplet.java:6081)
	at OpenBCI_GUI.playbackFileSelected(OpenBCI_GUI.java:24775)
	at OpenBCI_GUI.playbackFileSelected(OpenBCI_GUI.java:24727)
	... 21 more

retiutut, 2019-11-24:
There is an error in your playback file history! Please post the JSON here. @Sentdex

retiutut, 2019-11-24:
Alternatively, you can start the GUI and click Settings->"Clear All Settings" to start a new playback history.

closed 2019-11-24 by retiutut

retiutut, 2019-11-25:
I can add code to catch this exception in the future.

retiutut, 2019-11-26:
GUI 4.1.7 will delete any erroneous user playback history json file.