OpenBCI/OpenBCI_GUI #652: Can't detect Ganglion by bluetooth

amradel27, 2019-12-01:
## Problem

Failed to start dongle, I followed all the steps in the getting started guide, I tried troubleshooting the dongle as per this guide:
 http://brainwaves.io/wp/blog/2017/04/26/openbci-ganglion-windows-10-part-3-connecting/
yet still the same errors appear in the gui which are: 
code 411Error: serial port not open
failed to start dongle. it may not be plugged in

**I am using windows 10

gui version 4.1.7

Running standalone app

Ganglion

retiutut, 2019-12-01:
CSR dongle is no longer supported.

retiutut, 2019-12-01:
https://shop.openbci.com/collections/frontpage/products/ganglion-dongle?variant=15473352605768

closed 2019-12-01 by retiutut

amradel27, 2019-12-01:
and of course the older versions of the GUI that could work with the CSR dongle are not available ( how ignorant was that question from 1 to 10 )

retiutut, 2019-12-01:
Sorry about the CSR dongle, but I can assure you that the new Dongle is much more reliable and already compatible with current OS.