OpenBCI/OpenBCI_GUI #653: Impedance measurement doesn't work with wifi+cyton+daisy

bmartin427, 2019-12-04:
## Problem

Toggling the 'ohm' icons on and off seems to have no effect, other than a momentary jump in the signal, when interfacing over wifi.  This seems to be true whether using TCP, UDP, or UDPx3.  The same process works fine when using the serial interface, however this seems to require physically pulling the wifi shield out of the stack before this can be done.

See this forum post where the same issue was reported: https://openbci.com/forum/index.php?p=/discussion/2293/impedance-commands-not-working-through-wifi-shield-api-cyton-daisy

## Expected 

After toggling on the 'ohm' icon, a 31.2 Hz signal should be present in the time series and FFT plots.

## Operating System and Version

Ubuntu 18.04

## GUI Version

4.1.7

## Running standalone app

I've tried both the standalone app and invoking the hub/GUI directly from git checkouts.

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

Yes

## Console Log

`[1] imp pressed
[1] imp released
Imp[1] is on.
[DEFAULT]: Success: Impedance set.
[1] imp pressed
[1] imp released
Imp[1] is off.
[DEFAULT]: Success: Impedance set.
`

It seems the GUI thinks it's doing the correct thing at least.  Furthermore, I've also tried POSTing `z` commands directly to the wifi shield, and though the board reports `Success: Lead off set for 1`, I still don't get the expected impact on the data stream.

I strongly suspect this issue is not with the GUI, but somewhere further down the stack.  I am also actively investigating this issue myself.  I'm filing this because I felt this issue should be tracked somewhere, especially in light of the above forum post indicating other people have found the same issue.  I will open a new issue in the corresponding project if/when I track down the bug, and link from it here.

wjcroft, 2019-12-05:
Issue was logged previously on https://github.com/OpenBCI/OpenBCI_NodeJS_Wifi/issues/30 . As @retiutut mentioned there, the problem is likely not in the GUI itself.

bmartin427, 2019-12-05:
Update: so far, I've found that beginning streaming on wifi in the GUI somehow seems to change the LOFF register (0x04) on the ADS1299 from 0x02 to 0x00, which configures it for DC lead-off detection instead of 31.2Hz AC.  I'm still working to determine how that happens.

bmartin427, 2019-12-05:
I have further tracked this down to the sample rate command (e.g. `~6`) clobbering some ADS registers, which seems to be a firmware bug.  This is true whether or not the wifi board is present, but the GUI only attempts to change the sample rate when using the wifi interface.

Filed OpenBCI/OpenBCI_Cyton_Library#94, and closing here.

closed 2019-12-05 by bmartin427

wjcroft, 2019-12-05:
Brad, thanks very much for tracking this down.