OpenBCI/OpenBCI_GUI #656: In Linux BDF+ button does not create a BDF file

djan4848, 2019-12-11:
Hi, 
In Linux Ubuntu 18.04:
[image]

_drwxr-xr-x 2 djan djan    4096 dic  9 12:33 OpenBCISession_2019-12-09_12-12-31
-rw-r--r-- 1 djan djan 8298496 dic 11 11:14 temp.txt
djan@Base:~/Documents/OpenBCI_GUI/Recordings$ uname -a
Linux Base 4.15.0-72-generic #81-Ubuntu SMP Tue Nov 26 12:20:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux_

When trying to save a BDF+ file in GUI 4.1.7, the  file with extension ".bdf" is not created. 
Even if the file name is shown in the top of the GUI and the log, a "temp.txt" binary file is created on the folder "Recordings" in spite of a bdf file.
Even renaming it as temp.bdf does not work, for instance when trying to load such file on brainstorm.

That is the log:
djan@Base:~/DATA/Investigaciones/openBCI/OpenBCI_4.1.7$ ./OpenBCI_GUI
Console Log Started at Local Time: 2019-12-11_10-52-23
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Linux
TopNav: Internet Connection Not Available
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/djan/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
Hub: CheckHubInit: The GUI is connected to the Hub!
sourceList: click! 
setInterface: curInterface: -1
channel count set to 8
Setting Latency to 10000
Setting WiFi Internet Protocol to udpBurst
Setting WiFi style to dynamic
[DEFAULT]: Protocol Serial Selected for Cyton
setInterface: curInterface: 0
Setting sample rate for Cyton to 250Hz

sample rate set to: 250
[DEFAULT]: Transfer Protocol set to serial
serialList: click! 
[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0
channel count set to 16
[DEFAULT]: Output has been set to BioSemi Data Format (BDF+).
Radios_Config: system_status
[DEFAULT]: Attempting to open Serial/COM port: /dev/ttyUSB0
Radios_Config: connect_to_portName: Attempting to open serial port: /dev/ttyUSB0
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Success: System is Up
[SUCCESS]: Successfully connected to Cyton using /dev/ttyUSB0
[DEFAULT]: Autogenerated Cyton Session Name based on current date & time.
initButtonPressed: Calling initSystem()

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
Hub: processConnect: made it -- 144726 code: 200
Hub: setBoardType(): sending 'daisy -- 144726
Hub: processBoardType: success -- 144727
[SUCCESS]: The GUI is done initializing. Settings Loaded! Press "Start Data Stream" to start streaming! --  Cyton firmware v3.1.2
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
LoadGUISettings: Applied Widget 1 to Container 3
Networking Data Types || show num dropdowns = 6
LoadGUISettings: Applied Widget 3 to Container 9
LoadGUISettings: Applied Widget 10 to Container 4
Hub: sendCommand(String): sending '/3'
Cyton: setBoardMode to :DIGITAL
[DEFAULT]: Starting to read digital inputs on pin marked D11, D12, D13, D17 and D18
LoadGUISettings: Expert Mode Off
Hub: processCommand: success -- 145445
Networking: Protocol mode set to LSL
Networking Data Types || show num dropdowns = 6
Apply LSL Networking Mode
Ch 0, Active, x24, Normal, Include, On, Off
Ch 1, Active, x24, Normal, Include, On, Off
Ch 2, Active, x24, Normal, Include, On, Off
Ch 3, Active, x24, Normal, Include, On, Off
Ch 4, Active, x24, Normal, Include, On, Off
Ch 5, Active, x24, Normal, Include, On, Off
Ch 6, Active, x24, Normal, Include, On, Off
Ch 7, Active, x24, Normal, Include, On, Off
Ch 8, Active, x24, Normal, Include, On, Off
Ch 9, Active, x24, Normal, Include, On, Off
Ch 10, Active, x24, Normal, Include, On, Off
Ch 11, Active, x24, Normal, Include, On, Off
Ch 12, Active, x24, Normal, Include, On, Off
Ch 13, Active, x24, Normal, Include, On, Off
Ch 14, Active, x24, Normal, Include, On, Off
Ch 15, Active, x24, Normal, Include, On, Off
Success writing channel 0
Return code: 200
Success writing channel 1
Return code: 200
Success writing channel 2
Return code: 200
Success writing channel 3
Return code: 200
Success writing channel 4
Return code: 200
Success writing channel 5
Return code: 200
Success writing channel 6
Return code: 200
Success writing channel 7
Return code: 200
Success writing channel 8
Return code: 200
Success writing channel 9
Return code: 200
Success writing channel 10
Return code: 200
Success writing channel 11
Return code: 200
Success writing channel 12
Return code: 200
Success writing channel 13
Return code: 200
Success writing channel 14
Return code: 200
Success writing channel 15
New widget [0] selected for container...
Your cursor is in FFT Plot
New widget [1] selected for container...
Your cursor is in Networking
[DEFAULT]: Data stream started.
Cyton: startDataTransfer(): writing 'b' to the serial port...
Hub: sendCommand(char): sending 'b'
Hub: processCommand: success -- 183979
OpenBCI_GUI: openNewLogFile: opened BDF output file: /home/djan/Documents/OpenBCI_GUI/Recordings/OpenBCI-BDF-2019-12-11_10-55-24.bdf
Hub: sendCommand(String): sending '/0'
Cyton: setBoardMode to :DEFAULT
[DEFAULT]: Starting to read accelerometer
Hub: processCommand: success -- 205240
Hub: apparent sampleIndex jump from Serial data: 89 to  91.  Keeping packet. (1)
numPacketsDropped = 1
Interpolating dropped packets...
Hub: sendCommand(String): sending '/3'
Cyton: setBoardMode to :DIGITAL
[DEFAULT]: Starting to read digital inputs on pin marked D11, D12, D13, D17 and D18
Hub: processCommand: success -- 206760
Hub: apparent sampleIndex jump from Serial data: 24 to  27.  Keeping packet. (2)
numPacketsDropped = 2
Interpolating dropped packets...
[DEFAULT]: Attempting to use your default browser to launch: https://github.com/OpenBCI/OpenBCI_GUI/issues
[DEFAULT]: Attempting to use your default browser to launch: https://github.com/OpenBCI/OpenBCI_GUI/issues
Hub: apparent sampleIndex jump from Serial data: 85 to  88.  Keeping packet. (3)
numPacketsDropped = 2
Interpolating dropped packets...
 
And attached the screen captures 
[image]

jacobgmartin, 2019-12-11:
To get the BDF file output written you need to stop the session from the “system control panel”. 

I agree that it isn’t intuitive because it doesn’t seem to be documented.

djan4848, 2019-12-11:
Thank you for your help! , I've tried to close the GUI but not what you suggest.

closed 2019-12-11 by djan4848