OpenBCI/OpenBCI_GUI #657: OpenBCI GUI requires administrative rights to run

anthony-fontanez, 2019-12-12:
## Problem

OpenBCI GUI requires administrative rights to run

## Expected 

In a perfect world, this application would run without administrative rights

## Operating System and Version

Windows 10 Enterprise 1909 (OS Build 18383.476)

## GUI Version

4.1.7

## Running standalone app

Downloaded app

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

## Console Log

N/A

retiutut, 2019-12-14:
It has to read and write files to the local filesystem, and the .exe is signed.

cwhits, 2019-12-14:
@retiutut, can you open this issue back up? this would be a huge help for us too, since in its current state this app is impossible for us to deploy in a higher ed environment where professors are requesting the app and students don't have admin on the lab machines (for obvious reasons).

Ideally, we should be utilizing the user's AppData folder to store userdata/tempdata/etc, as the user has write access to their own profile. the main program files should be centrally located in a secured location within Program Files so all users can access it, but so that it cannot be modified except by someone with admin rights for the express purposes of maintaining/upgrading/uninstalling the application.

unfortunately, signing the exe doesn't help anything if it still requires admin rights to launch.

is there anything I can do to help move this issue forward or any other insight I can offer?

retiutut, 2019-12-14:
Thanks for adding more info.

retiutut, 2019-12-14:
We currently use the Documents folder on all OS, because it exists on all OS.

anthony-fontanez, 2019-12-14:
My apologies for not including additional information in the initial issue; I had logged it as I was leaving work for the day. I've got some experience in making various applications launch without admin rights, typically handled via additional filesystem and/or registry permissions, or by "shimming" it via MS Application Compatibility Toolkit. As for what is the underlying item that is making the exe require admin right though, that would be something specific in the code that I unfortunately don't have as much experience with. I mostly need to make software go for use in an education setting, so I'm usually dealing with scripting and building installers for mass deployment.

retiutut, 2019-12-14:
Asking @daniellasry to weigh in, as he as extensive experience with Windows OS.

daniellasry, 2019-12-14:
We should keep this bug open as it's a legitimate issue, or at least an opportunity for improvement.

However, I'm not sure if we'll be able to resolve it easily. I think admin permission are required to open a serial port. Some investigation needed.

One thing we're doing soon is getting rid of the HUB as separate process, which might get us closer to fixing this.

cfausn, 2019-12-16:
Also running into this issue for lab computers w/o admin rights... anyone have a temporary workaround to get the hub running outside of admin rights? Need it for next semester.

retiutut, 2020-01-16:
@anthony-fontanez @cwhits @cfausn Is a console log file made when attempting to run the GUI?

I think some lab computers can launch the GUI, but may be denied serial port or internet access to fetch current GUI version from Github.

anthony-fontanez, 2020-01-16:
@retiutut Simply attempting to launch the GUI triggers a UAC prompt for admin credentials.

retiutut, 2020-01-16:
@anthony-fontanez You think I can setup a test by making a Windows account w/o admin privilege? Any other Windows settings I should use on the new account?

Andrey1994, 2020-01-16:
requirement for admin priviligies specified in manifest file so the easiest possible test - remove it from manifest file, rebuild and try to run but more likely it will not work wo admin rights

retiutut, 2020-02-02:
I found this line in `gui.manifest` https://github.com/OpenBCI/OpenBCI_GUI/blob/9634ae767007eaa7b0f7d888e4f56ce9886ba17c/release_script/gui.manifest#L13

Andrey1994, 2020-02-03:
It's exactly what I am talking about, admin priviligies specified using lines like this one in manifest files. The first test should be removing lines like this one and trying to run GUI

retiutut, 2020-02-05:
This should be fixed now! There was an obvious requirement from admin rights in one of the manifest files, which has been changed to 
```
<requestedExecutionLevel level="asInvoker" uiAccess="false"></requestedExecutionLevel>
```
in the commit above.

retiutut, 2020-02-05:
To create a custom build of the GUI that stores data elsewhere on the computer, go to https://github.com/OpenBCI/OpenBCI_GUI/blob/9634ae767007eaa7b0f7d888e4f56ce9886ba17c/OpenBCI_GUI/SoftwareSettings.pde#L121

closed 2020-02-05 by retiutut