OpenBCI/OpenBCI_GUI #663: JS with cyton wifi shield problem

gerardklee, 2020-01-09:
Hi, I would appreciate any help! I will describe my problem below. I've attached my code and the image that shows the problem. In the image, I am referring to channel 1 (gray).
[image]
[image]

## Problem
I am working on project with Javascript and cyton wifi shield. As soon as I start streaming from the board using LSL and run my javascript file on the terminal (I have a mac), the signals completely stop on the GUI. I am assuming they are supposed to be continuosly moving while streaming.

## Expected 

Signals keep moving forward on the GUI while streaming the signals from the board to print the values on the console.

## Operating System and Version

**macOS**/Windows/Linux
version 10.14.6

## GUI Version
version 4.1.5

## Running standalone app

I don't understand what standalone app is

## Type of OpenBCI Board

**Cyton**/Cyton+Daisy/Ganglion

## Are you using a WiFi Shield?

**Yes**/No

## Console Log

No console log. Signals completely stop (meaning they don't move forward).

gerardklee, 2020-01-09:
I wanted to mention that streaming the data works perfectly fine with matlab. It is just in javascript, as soon as streaming begins, signals stop going.

retiutut, 2020-01-11:
It looks like you are trying to connect to the board twice at the same. You would need to accept an LSL stream in Javascript, or you can connect directly to the board using Javascript.

closed 2020-01-11 by retiutut