OpenBCI/OpenBCI_GUI #664: Cyton dongle used to connect to board but now times out

Deanis, 2020-01-09:
This morning everything was working fine, I was a able to get the data stream up and running.
I then added the daisy and extra 8 channels and now the bdongle won't connect to the board in either the 8 or 16 channel configuration.

The dongle shows up as com8 in device manager and the TXD and RXD lights flash when autoscan is pressed. The blue lights appear on both the dongle and board when powered up, howerever I am unable to make a connection using the OpenBCI GUI.
I don't understand why it stopped working.

Deanis, 2020-01-09:
This morning everything was working fine, I was a able to get the data stream up and running.
I then added the daisy and extra 8 channels and now the bdongle won't connect to the board in either the 8 or 16 channel configuration.

The dongle shows up as com8 in device manager and the TXD and RXD lights flash when autoscan is pressed. The blue lights appear on both the dongle and board when powered up, howerever I am unable to make a connection using the OpenBCI GUI.
I don't understand why it stopped working.

**Problem**
Cyton dongle does not connect to cyton baord even though both devices are on (blue lights) and the dongle appears as com8.

**Expected**
Cyton dongle connects to board and data is streamed

**Operating System and Version**
macOS/Windows/Linux

**GUI Version**
v4.1.7

**Running standalone app**
No

**Type of OpenBCI Board**
Cyton & Daisy

Are you using a WiFi Shield?
No

Console Log
N/A

retiutut, 2020-01-11:
Please provide a Console Log.

closed 2020-01-24 by retiutut