OpenBCI/OpenBCI_GUI #670: Having troubles interpreting the data from GUI

gerardklee, 2020-01-31:
BASIC INFO:
1) Using Wifi shiled
2) Using LSL to stream data
3) Had eyes closed
4) Only hooked up to 1st channel and 2nd channel
5) Exporting the "band power" data to the csv file using matlab
6) The trial ran for 3 minutes

I have been trying to figure out what the values represent and unfortunately could not. My goal is to analyze alpha waves and I am getting the "band power" data using lsl and exporting it to the csv file that looks as below: 
[image]
[image]

PROBLEMS:
1) I am thinking the columns represent the channels. I am not sure why the columns C-H have values as I am only hooked up to 1st channel and 2nd channel
2) I assumed that the middle row among the 5 rows with values would be the alpha band. However, as time went by, the alpha waves value stopped at 0.0001024 (you can see this in the second image). I expected the values would increase because I had my eyes closed, but then the values decreased and stopped at 0.0001024. 

I am having troubles to identify which rows/columns represent alpha here. Perhaps, my csv data is wrong..? I would appreciate any advice!

retiutut, 2020-01-31:
https://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIWidgets#networking

https://docs.google.com/document/d/e/2PACX-1vR_4DXPTh1nuiOwWKwIZN3NkGP3kRwpP4Hu6fQmy3jRAOaydOuEI1jket6V4V6PG4yIG15H1N7oFfdV/pub

closed 2020-01-31 by retiutut

retiutut, 2020-02-01:
This may be related to https://github.com/OpenBCI/OpenBCI_GUI/issues/644 depending on what you put in the Num Chan textfield in the Networking widget.

Have to close this because it doesn't fit a template and lacks important details.