OpenBCI/OpenBCI_GUI #705: Feature Request: GUI dark mode

projectmentor, 2020-03-27:
Is it possible to have an alternate css color scheme? Something like solarized dark for people with TBI and concussion it is difficult to visually process bright white background.  I just started here and would be willing to help once i get  up to speed on the documentations.

retiutut, 2020-03-27:
@projectmentor I am highly in favor since you have specifically requested an accommodation! Will add this to the next Software Team meeting. 💯

retiutut, 2020-03-31:
Currently, we have an alternate color scheme.

>To change to the alternate color scheme, use '{'.
https://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIDocs#expert-mode

So, we will either adding a third theme (dark mode), or replacing the alternate color scheme with dark mode.

Thank you for requesting this!

Dorson, 2020-04-09:
Have a look at this simple example. Unlimited color themes possible:

https://dorson.github.io/CSS-Dark-Mode-and-color-switch/

MaxJackson, 2020-10-08:
I'm thinking about picking this up and working on it! Would it make sense to have a 'Dark Mode' button available in the settings drop-down menu? If so, would it also make sense to move the 'alternate color scheme' configuration option over there too?

retiutut, 2020-10-08:
Sure, but behind the scenes I think we should make a "Style" class, where the mode can change and everywhere else says `style.getWidgetBackgroundColor()` etc.

retiutut, 2020-10-08:
@MaxJackson +1 priority bump. Thank you for supporting this feature request!

MaxJackson, 2020-10-11:
@retiutut I'm debating between 

- putting the toggle functionality in the ControlPanel and having it be a setting that the user can only toggle when initializing a session
- putting the toggle functionality in with the configOptions buttons and letting the user toggle it at any time

From what I can tell the first option will be much easier to implement, since the second option will require implementing a updateNavButtonsBasedOnColorScheme() analogue touching everything with a color we want to change.

Do you have any input/considerations before I go ahead and implement the first option?

MaxJackson, 2020-10-12:
@retiutut I added a Dark Mode / Light Mode toggle to the control panel and added dark mode functionality to every widget available in the synthetic data mode: check out https://github.com/MaxJackson/OpenBCI_GUI/tree/dark-mode-toggle to see where I'm going with this.

retiutut, 2020-10-12:
It's kind of all over the place. This is why I mentioned making a style class.

MaxJackson, 2020-10-12:
Great! Thanks for the feedback. I'm new to this project and to Processing in general, so I appreciate the guidance.

MaxJackson, 2020-10-12:
I'll do my best to implement your suggestion and keep you in the loop :)

retiutut, 2020-10-12:
I can try to get to this very soon. I can use this initial commit to get started.

retiutut, 2020-10-12:
@MaxJackson Can you share a screenshot of what you tried? That will also help.

MaxJackson, 2020-10-12:
Awesome, glad I could contribute :D

MaxJackson, 2020-10-12:
Here's how it renders:

[image]
[image]

retiutut, 2020-10-12:
Here is the commit for later:
https://github.com/MaxJackson/OpenBCI_GUI/commit/5ef61f777076279c38115b5e67b7cf847b4ce52e

retiutut, 2020-10-12:
Awesome! Thanks for getting this started. This is a great sign that it's time to knock out this issue.

MaxJackson, 2020-10-12:
Here's how I'm setting things up for now, generating arrays of colors or int(for black/white colors only)

[image]
[image]

Then I grab those colors in the widgets with (e.g.)

/// before the widget constructor
int thisWidgetFontColor = widgetFontColors.get(mainColorScheme);
color thisWidgetBackgroundColor = widgetBackgroundColors[mainColorScheme];
color thisGraphBackgroundColor = graphBackgroundColors[mainColorScheme];
int thisBoxLineColor = boxLineColors.get(mainColorScheme);
int thisGraphLineColor = graphLineColors.get(mainColorScheme);

// when initializing the plot 
plot.setAllFontProperties("Arial", thisWidgetFontColor, 14);
plot.setBgColor(thisWidgetBackgroundColor);
plot.setBoxBgColor(thisGraphBackgroundColor);
plot.setBoxLineColor(thisBoxLineColor);
plot.setGridLineColor(thisGraphLineColor);

MaxJackson, 2020-10-12:
I added some 

fill(thisWidgetBackgroundColor);
rect(x, y, w, h);

when the widget was more complex than the graph itself.

MaxJackson, 2020-10-12:
Consolidating the structure that contains the colors is certainly do-able; what also remains is
- evaluating the color scheme and checking consistency across widgets
- implementing dark mode in non-synthetic data mode widgets
- highlighting "LIGHT MODE" as the default when initializing the app for the first time

MaxJackson, 2020-10-12:
Thank you for your patience in reviewing and considering this! I'm happy to do the heavy lifting on this if you provide feedback and guidance. Let me know if you guys have a contributor slack/discord where there could be more real-time communication around this work.

MaxJackson, 2020-10-12:
Got it, that all sounds good. It sounds like I picked this up at a time of transition for a lot of the UI. Keep me in the loop, I'll be looking forward to collab-ing on this if it's something that makes sense to do :)

In the meantime I'll check out the button_obci->ControlP5.Button issue. Thanks again for the thoughtful feedback, I really appreciate the work you're putting in to make this project amazing.

retiutut, 2020-10-12:
>In the meantime I'll check out the button_obci->ControlP5.Button issue. Thanks again for the thoughtful feedback, I really appreciate the work you're putting in to make this project amazing.

Here is the a example of a self-contained cp5 instance in the SessionDataBox:
https://github.com/OpenBCI/OpenBCI_GUI/blob/44e89936e6530dea854df16251dd9a47b23af920/OpenBCI_GUI/ControlPanel.pde#L1585

You'll notice a callback listener is assigned close to the code where the button is created, inside the class where it is drawn. In this way, the code is "more together" rather than "spread out".

retiutut, 2020-11-21:
GUI v5.0.2 "The UI Update" now has almost the entire GUI's style defined in one place now (OpenBCI_GUI.pde). This was a significant effort to unify the UI objects into the ControlP5 library (which I updated, now v2.3.1 will show on GUI startup).

It is now possible to try and complete this ticket.

retiutut, 2021-05-07:
GUI v5.0.4 has almost all of the colors used consolidated as global variables in OpenBCI_GUI.pde.

**I would like to add this as a feature for GUI v5.1.0 on the grounds of increasing accessibility for all users.**

retiutut, 2025-03-31:
It's become obvious that this should be a higher priority for GUI v7 as it could help with readability and accessibility.

retiutut, 2025-03-31:
Use code from https://github.com/OpenBCI/OpenBCI_GUI/pull/1063

anaya33, 2026-03-03:
Hi @retiutut  so I am currently working on this feature (from dev branch), there are a few components that still need to be changed but here it is so far, after looking over some of previous code and implementation there is light mode based on #1248 + the open bci blue!

- I guess currently I am not sure how the channels component should be handled in terms of color in dark mode currently it is still grey, example:
  [image]

### Light Mode~
[image]

### Dark Mode~
[image]

### Legacy Mode~
[image]