OpenBCI/OpenBCI_GUI #709: sound tones beats and more

professorderekdecker, 2020-04-05:
I have the Ganglion and wish to practice changing brain states through meditation, concentration, etc.  However, in order to have biofeedback, I currently have to look at the OpenBCI GUI screen. This is not conducive to meditative relaxation. I ask that the 5 brain wave ranges each be assigned a note in a chord (the major 7th or other melodious combination). With eyes closed, I'll be able to listen to the notes and learn how best to change them (aka brain states). It may also be useful to impose a modulation (slower modulation on the lower notes for lower brain wave ranges). That way, if your mind drifts away from recognizing notes, it can hone in on the modulation. Also, for those interested in combining this with binaural beats, we may need a binaural beats generator and mixer (to combine the biofeedback and beats stimulation).

professorderekdecker, 2020-04-05:
By the way, I did try neuromore but was unable to get the Ganglion to connect. If another sound generating biofeedback device or software works with the Ganglion, then please let me know. Thank you.

retiutut, 2020-04-06:
You can try Brainflow + TimeFlux: https://doc.timeflux.io/latest/general/about.html

This is scheduled to be added to the OpenBCI docs.

Marking this issue as a new feature request! 😄

retiutut, 2020-05-01:
More details here from similar issue: https://github.com/OpenBCI/OpenBCI_GUI/issues/745

daniellasry, 2020-05-01:
Hey Derek, it sounds like the ideal scenario for you would be to get the Ganglion working with Neuromore.

Unfortunately I'm not very familiar with neuromore. It seems like they added Cyton support at some point, but not for Ganglion. Am I correct?

We have a fresh new API for OpenBCI Boards: brainflow. It can be used in Python, Java, R, C++, C#, Matlab, Julia. I see that neuromore is open source and based in C++. Maybe the neuromore devs would be open to integrating brainflow. IF they did, they would be adding support for all OpenBCI boards at once.

[brainflow source](https://github.com/OpenBCI/brainflow)
[documentation](https://brainflow.readthedocs.io/en/stable/)

rakeshbci, 2020-09-23:
You can connect ganglion and cyton to neuromore using OSC proctocol. Stream the data from GUI to a python script and stream each channel to a specific OSC address.

professorderekdecker, 2020-09-24:
Is there a video or website or pdf or anything that gives more detail about how to do this?

Thank you,
Derek

professorderekdecker, 2020-09-28:
I downloaded and tried neuromore again. The same behavior occurs. It will briefly connect to the Ganglion, then disconnect and won't reconnect. Very frustrating!

What is "OSC proctocol. Stream the data from GUI to a python script and stream each channel to a specific OSC address."? I have no idea what this means. How do I learn how to do this? Are there any examples I can download? Who has done this and how do I connect with them?

Derek
________________________________
From: Derek Decker <derekdecker@hotmail.com>
Sent: Wednesday, September 23, 2020 7:01 PM
To: OpenBCI/OpenBCI_GUI <reply@reply.github.com>
Subject: Re: [OpenBCI/OpenBCI_GUI] Feature Request: Simple Neurofeedback - assign tones to Band Power widget (#709)

Is there a video or website or pdf or anything that gives more detail about how to do this?

Thank you,
Derek

PR #982 (2021-07-31): Auditory Neurofeedback + Focus - 5.0.6
Fixes #709 

- Use the Metric to control all notes at once, or use Band Powers to control certain notes of the chord.
- Carefully positioned UI for Auditory Feedback (currently two buttons)
- Chords notes have varying length. When looped, it sounds continuous.
- Samples are cached in memory asynchronously on app start to improve audio quality and playback
PR #985 (2021-07-31): GUI 5.0.6 - July 2021
# v5.0.6

### Improvements
* Add Auditory Feedback to the Focus Widget Fixes #709

### Bug Fixes
* Fix drawing error in Control Panel WiFi Shield static IP Textfield
* Accomodate high-dpi screens Fixes #968
* Add Arduino Focus Fan example to networking test kit on GitHub repo
* Allow synthetic square wave expert mode keyboard shortcut for Cyton and Ganglion Fixes #976
closed 2021-07-31 by retiutut