OpenBCI/OpenBCI_GUI #718: Incorporating Event markers from /Open Sesame/Psycho py/uperlab into data stream

Jarrod-J, 2020-04-15:
## Problem

I am interested in using OpenBCI  hardware for ERP.   I would like to know how easy it is to incorporate event markers into the data stream  using the GUI . I currently present my experiments using software such as Opensesame    which send a  numberic event marker  via  a serial COM port.

What I would like to know is  how do I set the OpenBCI GUI  to accept the codes via a specified COM port.  

Also -  will these event codes appear in a particular channel or be in a separate part of the data?

I would appreciate help and guidence on this.

Jarrod Hollis

## Operating System and Version   Widows  10  64 bit

## GUI Version 4.2.0  Standalone version of app

retiutut, 2020-04-17:
This is a weird request, and it seems like an isolated case for a particular purpose. You are welcome to edit the GUI code as you see fit, since it is under the MIT license.

You can look at https://processing.org/reference/libraries/serial/Serial_read_.html to get you started.

Hope this helps! Closing this issue since it is an isolated feature request.

retiutut, 2020-04-17:
Normally, event markers are handled and saved to file by using the Analog Pins on the Cyton.

closed 2020-04-17 by retiutut

Jarrod-J, 2020-04-17:
What a strange reply?   I was interested in finding out how   events operates  BEFORE  purchasing  BCI  and I was  told to post on here.    The point is that I need to know how events are implemented.  I  will stick to EMOTIV and  I will just purchase a different amplifier

retiutut, 2020-04-17:
Event markers are injected using Analog/Digital Pins on the Cyton. Your original post asks for a new system of handling event markers.

retiutut, 2020-04-17:
OpenBCI software is free, data is free, and users are free to modify the program to their needs. Emotiv will likely charge for these features.