OpenBCI/OpenBCI_GUI #745: Auditory EEG biofeedback

professorderekdecker, 2020-05-01:
I'm always frustrated when Ganglion won't connect to neuromore. Since that may not get solved anytime soon, PLEASE add sounds for biofeedback in the current GUI.

Each brain state (ranges of frequencies) should be assigned a singe note in a melodic chord. Hearing the notes allows us to keep our eyes closed during EEG monitored meditation and hopefully, to help us guide ourselves to the brain states we want using this biofeedback. I find having to open my eyes to look at the computer screen waveforms to be very distracting, and not conducive to more relaxed states of consciousness.

retiutut, 2020-05-01:
It makes sense and sounds like a good idea, but we already have an issue that you made requesting something similar:

https://github.com/OpenBCI/OpenBCI_GUI/issues/709

We are trying to release GUI v5 in the very near future, and this would only be added after we are done with the core of the program. Just keeping you up-to-date with the status of GUI development.

closed 2020-05-01 by retiutut

professorderekdecker, 2020-05-01:
Thanks for the update. I could not find my prior issue so I started a new. Any idea when this will get implemented (weeks, months, years)? Also, has Gangion/Neuromore been fixed?

Thanks,
Derek

#709<https://github.com/OpenBCI/OpenBCI_GUI/issues/709>