OpenBCI/OpenBCI_GUI #766: Unable to establish link to Hub

Dream-build, 2020-05-12:
## Problem

Unable to establish link to Hub. It worked and started to stream data from the board for the first testing, but failed in the following test.

## Expected 

Let my gui to establish link to the newest hub

## Operating System and Version

Windows 10

## GUI Version

4.2.0 64 bit

## Running standalone app

Downloaded app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

I also posted a discussion in the openbci forum
https://openbci.com/forum/index.php?p=/discussion/2553/bug-troubleshooting-needed-openbcihub-fatal-error-listen-eaddrinuse-127-0-1-10996#latest

retiutut, 2020-05-12:
The Hub has/is being removed from the GUI for 5.0. Closing this issue. The error you are seeing means that the Hub is already running, and you have tried to open the same port twice.

closed 2020-05-12 by retiutut

Dream-build, 2020-05-13:
> The Hub has/is being removed from the GUI for 5.0. Closing this issue. The error you are seeing means that the Hub is already running, and you have tried to open the same port twice.

Sorry sir, but even if I only open the gui, the gui is still not working  and can still not find the hub. Did I just found a bug? @retiutut

daniellasry, 2020-05-13:
Hey @Dream-build, if you look in your app tray, you might see a dark icon for the OpenBCI HUB. Right click -> Quit to close the app. Then try opening the gui again.

If you can't find it, try logging out and logging in again.

Dream-build, 2020-05-13:
@daniellasry   @retiutut Thank for the help sir, but I tried for many times and it still seems not working. Here is the record of the log. Can you help me to find the problem?

No library found for com.sun.jna
No library found for com.sun.jna
No library found for com.sun.jna
No library found for com.sun.jna
Console Log Started at Local Time: 2020-05-13_12-53-09
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
Launching application from local data dir
OpenBCI_GUI: hubStart: OS Detected: Windows
TopNav: Internet Connection Not Available
Local GUI Version: 420
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "C:\Users\asus\Documents\OpenBCI_GUI\Settings\UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
OpenBCI_GUI::Setup: Setup is complete!
Instantiating hub object...
sourceList: click! 
[ERROR]: Unable to establish link to Hub. LIVE functionality will be disabled.
ControlEvent: Hub error
sourceList: click! 
[ERROR]: Unable to establish link to Hub. LIVE functionality will be disabled.
ControlEvent: Hub error

retiutut, 2020-05-13:
The Hub has already been completely removed from the GUI. Also, this is a duplicated issue of a post in the forums.