OpenBCI/OpenBCI_GUI #775: LSL manual timestamping interferes with LSL clock_offset correction

cboulay, 2020-05-21:
## Problem

Hi. I'm one of the LSL maintainers. I'm trying to help a user with a problem. They are unable to synchronize OpenBCI streams with those from different sources when loaded xdf files saved in LabRecorder.

The xdf importer adjusts the timestamps in each stream to a common clock: lsl's local_clock on the computer that was running LabRecorder. The clock_offsets are calculated online every ~5 seconds using the LabRecorder computer's and the **remote computer's lsl::local_clock**. When the remote stream's timestamps also uses local_clock, then synchronizing the streams on data import 'just works'.

When the remote streams timestamps are set manually, as you might want to do if you had information from the hardware about sample_acq_time which you know to be superior to the default sample_push_time, then it's important to convert sample_acq_time into the same base clock as lsl_local_clock. 
If a different base clock is used then post-hoc synchronization of different LSL streams is impossible.

FYI, [lsl's clock is](https://github.com/sccn/liblsl/blob/a6966233cfd4d2fca52a8f75800d73a624cd4a9f/src/common.cpp#L22-L26) `std::chrono::high_resolution_clock::now().time_since_epoch()`.

Another important details is that the timestamps should be in seconds. When using `push_chunk`, timestamps for samples after the first sample in a chunk are auto-incremented based on the stream's nominal_rate as +1/rate **seconds**. If millisecond timestamps are used then the first timestamp in a chunk will be in milliseconds but subsequent timestamps in that chunk will be incremented in seconds. I expect this would lead to a situation where a a chunk's first timestamp is earlier than the last timestamp of the preceding chunk.

## Expected 

As it appears the manual timestamps are simply the time at sample push and do not make use of any hardware timing info, there's no benefit to supplying the timestamps manually. It would be simpler to not provide timestamps at all and let LSL timestamp them for you. This will enable post-hoc synchronization of different streams and fix the chunk timestamping error above.

The downside to this is that it might break compatibility with other software you might have that expects the LSL timestamps to be `System.currentTimeMillis()`. For example, if any of your example Matlab or Python scripts for reading LSL streams expects timestamps in msec then these will have to be adjusted.

retiutut, 2020-05-21:
Thank you for a thorough breakdown of what is happening. We will review this carefully.

LJGz, 2020-05-22:
I am the user, and happy to help providing example data or testing.

retiutut, 2020-05-22:
So we should revert 
```
 outlet_data.push_chunk(dataToSend, System.currentTimeMillis());`

TO

outlet_data.push_chunk(dataToSend);`
```
?

cboulay, 2020-05-22:
That should be all that is required, though I'm not sure how this interacts with #530.

retiutut, 2020-05-22:
>I am getting LSL streams from openbci gui to pylsl and trying to get an absolute timestamp with the correct epoch. But since the timestamp sent by Gui is relative, matching it with any other external timeseries becomes hard. Anyway to avoid this simply by adding to the timestamp the current time from the gui perhaps?

>If a different base clock is used then post-hoc synchronization of different LSL streams is impossible.

I'd like to reconcile both issues. So the goal in both cases is to sync multiple LSL streams coming from different sources? Example: 1 stream from GUI, another stream from a 3rd party Python script.

cboulay, 2020-05-22:
Right. #530 was not necessary (and in fact was incorrect).

If all streams are on the same PC (here I'm using PC for all types of computers, even Mac and mobile), then synchronization is trivial.

If streams are on different PCs, then the raw timestamps by default are those returned by `inlet.pull_chunk` are those generated by the local_clock on the source PC. What local_clock uses, or more precisely what `high_resolution_clock` uses, for `epoch` is unix time (1970-01-01 midnight) on most platforms, but on MacOS (at least older versions) its `epoch` is when the PC was last rebooted. 
In #530 @compmonks was using MacOS so this is likely the problem. Even if all computers used Unix time as t=0, it's still bad to rely on this clock because computer's are easily off by a few seconds relative to each other and there's also timezone issues that come into play. The solution is to use the aforementioned `clock_offsets` that are calculated by an inlet.

While the default during loading xdf files is to use those clock_offsets to correct timestamps, that is not the default for online pulling from an outlet e.g. using pylsl. The solution to #530 was to set the postprocessing flags: 
https://github.com/labstreaminglayer/liblsl-Python/blob/master/pylsl/pylsl.py#L80-L86

@compmonks should instead use something like the following:
`my_inlet = StreamInlet(stream_info, processing_flags=pylsl.proc_ALL)`

So why don't we synchronize timestamps by default? Probably because `processing_flags` was a relatively recent addition to pylsl and we wanted to maintain backward compatibility. Maybe we should raise a warning whenever post processing is not used and clock_offsets are found to be > eps.

retiutut, 2020-06-02:
From @Andrey1994 :
>timestamps collected on the SDK are much more precise and dont introduce additional delays like here...

He is referring to the timestamps managed by the BrainFlow SDK (data acquisition). @cboulay Would this be helpful? Or should we send data using `push_chunk(data)` as discussed on this thread?

closed 2020-06-02 by retiutut

cboulay, 2020-06-02:
I'm not familiar with the BrainFlow SDK and I don't know where to begin looking to see how they manage timestamps. But I believe it's possible to get more precise timestamps. (I also expect they don't fully understand what LSL does under the hood so the difference might not be as great as you expect).

You can certainly use more precise timestamps, and I encourage you to do so if you have them, but they have to be converted to the LSL clock first. This is usually as simple as: `lsl_ts = precise_ts * scale + offset`. Note that if `precise_ts` is coming from a source different to what the LSL clock is coming from, then `scale` and `offset` might drift over time.

varjak, 2022-05-11:
I have a python script that sends two streams (data + markers) through LSL with manual timestamps to Labrecorder, all with one (windows) computer. When I load the .xdf file in MATLAB (with load_xdf.m), the data timestamps **sometimes** change and the data become shifted from the markers.

I found that if I call `load_xdf('file.xdf', 'HandleClockSynchronization', 0)`,  the timestamps are loaded without change, and data + markers are alligned. Should I just do this, or are there any disavantages of not doing the load_xdf clock synchronization step?

From this thread, I got that I could also:
- convert the manual timesptamps to the local LSL clock first, I assume in the python script, before they are pushed. How would I do that?
- not push the values (data / markers) with the timestamps, so do `push_sample(array_of_values)` instead of  `push_sample(array_of_values, array_of_timestamps)`. But as I know the sampling frequency of the array_of_values, I think I would prefer to manually generate the array_of_timestamps, without relying at the actual times they were sent.

cboulay, 2022-05-11:
Hi @varjak , is this question related to OpenBCI? If not then may I ask that you bring this question to one of the LSL repos (sccn/labstreaminglayer or sccn/liblsl)? Or you can join our Slack channel (link in sccn/labstreaminglayer README).

PR #786 (2020-06-02): Let LSL manage its own timestamps
Closes #775