OpenBCI/OpenBCI_GUI #804: wifi shield python script to load accelerometer data

tabh, 2020-06-11:
## Problem
I cant read accelerometer data from wifi shield with python script 
Explain the problem
Hi,
I am trying to read 8 channel EEG+ 3 axes Acc data from wifi shield with python script. 

I am using the following script:    
But there is no data att sample.aux_data

**from pyOpenBCI import OpenBCICyton
from openbci import wifi as bci

def print_raw(sample):
    print(sample.channel_data)
    print(sample.aux_data)

shield_name = 'OpenBCI-6F92'

board = bci.OpenBCIWiFi(ip_address='192.168.4.1', shield_name=shield_name, log=True, high_speed=False, num_channels=16)
board.start_streaming(print_raw)
board.loop()**

## Expected 

In a perfect world, what do you expect to happen.
I expect to get both EEg and Aux data printed
## Operating System and Version
Windows 10.
macOS/Windows/Linux

## GUI Version

The version is displayed on startup.

## Running standalone app

Are you running the downloaded app or are you running from Processing or Visual Studio Code?

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion
Cyton board 
## Are you using a WiFi Shield?
Yes
Yes/No

## Console Log

Paste any relevant text from the console window here

tabh, 2020-06-11:
I will really appreciate a help with a simple script to read data in python

retiutut, 2020-06-11:
pyOpenBCI has been deprecated. Please try using BrainFlow instead: https://brainflow.readthedocs.io/en/stable/Examples.html#python

closed 2020-06-11 by retiutut