OpenBCI/OpenBCI_GUI #805: Non-Smooth Connection； Ganglion ー BLED112 Dongle

Otemo, 2020-06-13:
## Problem
Every time I try to connect to the ganglion board via Bluetooth (BLED112 dongle) using the GUI, the COM port on the BLED112 dongle is updated. Windows informs me via a sound that the BLED112 Dongle COM port is refreshing.

Also, even if I stop the data streaming of the GUI and stop the session, the connection with the Ganglion board is still maintained. 
The blue diode on the Ganglion board does not flash and remains lit.

Although not smooth, we can make these connections and record data.

## Expected 
COM port doesn't refresh when connecting
Blue diode blinks when the connection is stopped

## Operating System and Version
Windows10

## GUI Version
v5.0.0-beta.0

## Running standalone app
we are running the downloaded app  and  from Processing

## Type of OpenBCI Board
Ganglion

## Are you using a WiFi Shield?
No

## Console Log

Paste any relevant text from the console window here

Otemo, 2020-06-13:
# Console Log
Console Log Started at Local Time: 2020-06-13_15-52-30
Screen Resolution: 1366 X 768
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
TopNav: Internet Connection Not Available
Local GUI Version: 500
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 4
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
bleList: click! 
bleList: click! 
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
SHUTDOWN HOOK

retiutut, 2020-06-15:
@Otemo Thank you for noticing this and sharing details! I will bring this up at this week's software meeting.

retiutut, 2020-06-29:
@Otemo Does the GUI show these errors every time you try Ganglion + BLED112?

Can you connect at all using GUI v5 on your Windows10 computer?

Otemo, 2020-06-29:
Hi @retiutut 
Yes, it does. every time i try Ganglion + BLED112.

However, Cyton board works fine. 
I use GUI v5 on my Win 10 laptop.

retiutut, 2020-06-29:
Does GUI v4.2.0 work with your Ganglion + BLED112?

Otemo, 2020-06-29:
Yes, it works, but we have detected random spikes  [>forum](https://openbci.com/forum/index.php?p=/discussion/2470/random-spikes-with-gui-bled112-but-not-bioera-resolved#latest)

Andrey1994, 2020-06-29:
I still dont understand what the issue is. Sound during init session is not a bug and expected.
And as I undersant from "Although not smooth, we can make these connections and record data." it works

Otemo, 2020-06-29:
Hi @Andrey1994,

we always appreciate  your support!

I've uploaded the video when the GUI is successfully connected. Is this behavior as you expected? 
> [GUI_5.0_on_win10](https://drive.google.com/file/d/10jNsH-dCQrmKgi2g79X5ZV6fwiBjow8w/view?usp=sharing)

I sometimes experience unstable connections. 
However, when it connects successfully, the GUI connects as as I showed in vod. 

I hope it works as you expected.

Andrey1994, 2020-06-29:
There are 2 libraries which work with this COM port - GUI Helpers to discover devices in the beginning and BrainFlow to do everything else. To get device to the known state both of them call 
```
ble_cmd_system_reset()
```
from BGLib(https://www.silabs.com/community/wireless/bluetooth/forum.topic.html/resetting_bled112do-TrHm) this method leads to the sound you hear.

I think its safe and correct to force known state for device. Downside of it - sound you hear and small delay. In BrainFlow I will definitely keep this reset command. In GUI Helpers maybe it can be removed

Otemo, 2020-06-29:
Hi @Andrey1994,

Thank you for your polite explanation!

I believe that these are not bug, normal phenomena.
(I am concerned that the connection is unstable....)

From my point of view as a user, it's okay to have a sound and a slight delay when connecting.

Andrey1994, 2020-06-29:
what do you mean by unstable connection? I didnt see smth like that on video you shared

Otemo, 2020-06-29:
hi @Andrey1994,  
we can occasionally observe the following phenomena.

-*-*-*-*-*-*-*-*-*--*-*-*-*-*-*-*-*-*--*-*-*-*-*-*-*-*-*-
Console Log Started at Local Time: 2020-06-29_12-13-41
Screen Resolution: 1536 X 864
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
TopNav: Internet Connection Not Available
Local GUI Version: 500
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 4
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
bleList: click! 
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
[DEFAULT]: No BLE device selected. Please select your Ganglion device and retry system initiation.
[DEFAULT]: BLE Devices Refreshing
ControlPanel: Found BLED112 Dongle on COM port: COM3
openbci_gui_helpers.GanglionError: Error in scan for ganglions:PORT_OPEN_ERROR
	at openbci_gui_helpers.GUIHelper.scan_for_ganglions(GUIHelper.java:69)
	at OpenBCI_GUI$BLEBox$1.run(OpenBCI_GUI.java:5502)
ConsoleLog: Console closed!
SHUTDOWN HOOK

Andrey1994, 2020-06-29:
is it during autodiscovery? how many times do you click search button?

Otemo, 2020-06-29:
we did not press the "search" button while the GUI was automatically discovering the device. 
The ganglion board may not be detected automatically, even if we wait for the GUI to detect it. 
At this time, we pressed the "start search" button approximately once every 10 seconds.

Otemo, 2020-06-29:
hi @Andrey1994,
As you pointed out, the connection instability may be due to my poor operation. 

We could have pressed press the "start search" button while the GUI was searching for devices.
>[GUI_5.0_connection_failed](https://drive.google.com/file/d/1onR848NuyGk6Bp-HFhSuVHAbSTGrwIjE/view?usp=sharing)

If these are the cause, we apologize for making you worry.

retiutut, 2020-08-13:
@Otemo Can you please try https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v5.0.0?

Otemo, 2020-08-13:
Yes. This issue has been resolved. 
Thank you for your support！

PR #882 (2020-09-21): Development 5.0.1 -> Master
# v5.0.1

### Improvements
* Add ability to save and load hardware settings
* Add configurable gain behaviour
* Add custom vertical scale UI to Time Series

### Bug Fixes
* Fix #805
* Covert GUI v4 sample data to GUI v5 format #830
* Display GUI version in title bar, along with FPS

### Bug Fixes
* Check internet connection to Github using a timeout, so the app doesn't stall
closed 2020-08-13 by retiutut