OpenBCI/OpenBCI_GUI #816: OpenBCI Hub 5.0.0 [ISSUE]

john-mpelkas, 2020-06-24:
## Problem

I am trying to run the OpenBCI_GUI on my Ubuntu 20.04 LTS machine. The issue is that the GUI will not run. I am able to run the older versions of this OpenBCI_GUI (4.7 I believe is the previous version?). When I run `./OpenBCI_GUI` the stage opens up but the scenes are black. I will attach the error message I get in the console to this ticket. I have looked around the web for answers but have not been able to get a clear reason why this error is occurring.

## Expected 

I would expect that the GUI would open with the options of boards to choose from to connect to.

## Operating System and Version

Linux (Ubuntu 20.04 LTS)

## GUI Version

OpenBCI_GUI (5.0.0 Beta)

## Running standalone app

I am running the application that I downloaded from the openBCI download page.

## Type of OpenBCI Board

Cyton 

## Are you using a WiFi Shield?

No

## Console Log
java.lang.RuntimeException: Profile GL4bc is not available on X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x7f3dec016da0, owner true, ResourceToolkitLock[obj 0x432e6e2e, isOwner true, <4711e293, 2136415>[count 1, qsz 0, owner <main-FPSAWTAnimator#00-Timer0>]]], but: [GLProfile[GLES1/GLES1.hw], GLProfile[GLES2/GLES3.hw], GLProfile[GL2ES1/GLES1.hw], GLProfile[GL4ES3/GL4.hw], GLProfile[GL2ES2/GL4.hw], GLProfile[GL4/GL4.hw], GLProfile[GLES3/GLES3.hw], GLProfile[GL4/GL4.hw], GLProfile[GL3/GL4.hw], GLProfile[GL2GL3/GL4.hw]]
at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:410)
at java.lang.Thread.run(Thread.java:748)
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f3dec0012b0, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f3dec016da0, refCount 1, unCloseable false]

[image]

retiutut, 2020-06-25:
@john-mpelkas As I understand, this is a specific message relating to running Linux, using Processing (this is what helps us draw the GUI), and the graphic card(s) on your computer.

Check out this comment on an open issue: https://github.com/processing/processing/issues/5476#issuecomment-448984413

I'm not sure how comfortable you are editing a few lines of code, but this might be worth a try, and I can help! To do this, you would need to download Processing and run the GUI from the code. This is described in the main GUI doc: 

http://docs.openbci.com/docs/06Software/01-OpenBCISoftware/GUIDocs#running-the-openbci-gui-from-the-processing-ide

john-mpelkas, 2020-06-26:
Thank you @retiutut! 
The software seems to be working fine now. Would it be possible to turn this file back into an executable so I could just run it via `./OpenBCI_GUI` or will I have to open Processing 3 every time?

retiutut, 2020-06-26:
Oh wow! Nice to know! I will share this information with the Software Team.

If the fix did indeed work for you, you can make your own executable using Processing for now.

john-mpelkas, 2020-06-26:
Alright, I'll look into creating an executable using Processing. As long as the main OpenBCI gui is working I am sure I can manage.

Thank you again! =]

closed 2020-06-26 by john-mpelkas

daniellasry, 2020-06-28:
@retiutut let's apply this fix to our code, does not look like it negatively affects anything in windows.

retiutut, 2020-06-29:
@john-mpelkas We are going to roll out this fix for everyone! Thanks for making this issue. 👍