OpenBCI/OpenBCI_GUI #830: OpenBCI GUI v5.0 freeze at "Attempt to start session" upon Sample Data opening on two different MacOs Sierra. v4.2 work.

clemriedel, 2020-07-17:
## Problem

OpenBCI GUI v5.0 freeze at "Attempt to start session" upon Sample Data opening on MacOs Sierra

## Expected 

To start the GUI. v4.2 worked for me

## Operating System and Version

macOS sierra 10.12.6

## GUI Version

5.0

## Running standalone app

Downloaded App

## Type of OpenBCI Board

No board, just playback

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

retiutut, 2020-07-18:
I think this is because we need to either convert the sample data or make new recordings that are compatible with GUI v5.

Thank you for making an issue!

retiutut, 2020-08-03:
A fix for this issue has just been merged. Thank you for making this issue!

PR #843 (2020-08-03): Convert gui v4 sample data to v5 - Fixes #830
- [x] Fixes #830
- [x] Write some code to make sure we delete GUI v4 sample data files in Documents folder, and place new files there 
- [x] minor refactoring of related resources
commit af011bef (2020-08-03): Merge pull request #843 from retiutut/gui-v5-sampledata-update

Convert gui v4 sample data to v5 - Fixes #830
commit f28a4db0 (2020-08-03): Convert gui v4 sample data to v5 - Fixes #830
closed 2020-08-03 by retiutut