OpenBCI/OpenBCI_GUI #837: Channels 9-16 don't work with Cyton/Daisy/Wifi

nateGeorge, 2020-07-23:
## Problem

When using the GUI with Cyton and the Daisy extension along with the wifi board, the data from channels 9-16 don't show up.  I've confirmed data from these sensors works when using a different software (https://github.com/bmartin427/bci_logger).  Channels 9-16 are flat lines with 0 nV.

## Expected 

The data from channels 9-16 should show up.

## Operating System and Version

I have tried on Windows 10 and Ubuntu 16.

## GUI Version

5.0.0-beta.0

## Running standalone app

standalone

## Type of OpenBCI Board

Cyton+Daily

## Are you using a WiFi Shield?

Yes

## Console Log

```
Console Log Started at Local Time: 2020-07-23_17-15-59
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
TopNav: Internet Connection Successful
Local Version: 500, Latest Version: 500
GUI is up to date!
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/nate/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
null
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 8
[DEFAULT]: Wifi Devices Refreshing
wifiList: click! 
[DEFAULT]: Selected WiFi Board: OpenBCI-51AA, WiFi IP Address: 10.0.0.220
Channel count set to 16
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Sending config string to board: ~5
Channel count set to 16
InitSettings: Saving Default Settings to file!
SoftwareSettings: 3 active widgets saved!
LoadGUISettings: Applied Widget 0 to Container 4
LoadGUISettings: Applied Widget 1 to Container 3
LoadGUISettings: Applied Widget 2 to Container 9
LoadGUISettings: Expert Mode Off
Networking: Protocol mode set to Serial. Stopping network
Networking Data Types || show num dropdowns = 13
Apply Serial Networking Mode
[DEFAULT]: Data stream started.
OpenBCI_GUI: openNewLogFile: opened ODF output file: /home/nate/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2020-07-23_17-16-01/OpenBCI-RAW-2020-07-23_17-33-47.txt
Brainflow start streaming
[DEFAULT]: Data stream stopped.
Brainflow stop streaming
ConsoleLog: Copying console log to clipboard!

```

Here is an image:
[image]

I'm not sure if this is the reason for this, but it looks like from the logs it's sending a config string of 5 for cyton+wifi when it should be setting 6 for cyton+daisy+wifi (https://brainflow.readthedocs.io/en/stable/UserAPI.html), if that config string is the board id.

daniellasry, 2020-07-24:
Thanks @nateGeorge , I am able to reproduce the issue. The "~5" command you see is setting the sample rate, not the board id. It look like the correct board ID is used, but there is a bug causing the channels to come out as zero. I'll let the team know.

Daniel

retiutut, 2020-07-27:
@daniellasry looks like y'all were able to merge some code to resolve this issue. Can this be closed now?

retiutut, 2020-07-27:
@nateGeorge Thank you for making an issue, this is a great help while the GUI is in beta. We hope to make a full/stable release soon.

daniellasry, 2020-08-05:
@nateGeorge this is fixed in the latest 5.0.0 stable release. Let us know if you still have issues.

closed 2020-08-05 by daniellasry

kvkoshmak, 2020-08-12:
Good day everyone, I'm having the same issue in Linux OpenBCI GUI. Update button in GUI shows that I'm using the newest version. Was the new update released or it is necessary to run some process manually?

nateGeorge, 2020-08-13:
Nice, thanks @daniellasry !  Finally had a chance to test it today and it's working on both Linux (Ubuntu 18) and Windows 10.  @kvkoshmak , you probably aren't using the latest version.  It should say 'August' when starting up and when downloading from the OpenBCI downloads page.

kvkoshmak, 2020-08-14:
Thanks @nateGeorge manually download and replace the GUI worked. Update button from GUI didn't make any changes for some reason.

nateGeorge, 2020-09-27:
> @nateGeorge this is fixed in the latest 5.0.0 stable release. Let us know if you still have issues.

It would've been nice to reference this issue in the commits so we could see what changed where.