OpenBCI/OpenBCI_GUI #844: java.lang.UnsatisfiedLinkError: liblog.so

jacobgmartin, 2020-07-28:
## Problem

I copied all the libraries from OpenBCI_GUI (latest development version) over to the ~/sketchbook/libraries directory for Processing 4 alpha1.

Yet, I seem to have two missing libraries "liblog.so" and "processing.awt.PSurfaceAWT"

I can't seem to find any information about this error or this library on the internet.  

I'm on Gentoo.  Does anyone have any advice? Should these libraries be included in the libraries sketchbook subfolder?

## Expected 

No libraries missing.

## Operating System and Version

Linux  5.4.38-gentoo-x86_64 #2 SMP Fri May 15 04:51:57 EDT 2020 x86_64 Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz GenuineIntel GNU/Linux

## GUI Version

5.0.0-beta2

## Running standalone app

From processing.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

**No library found for processing.awt.PSurfaceAWT**
Console Log Started at Local Time: 2020-07-28_18-09-19
Screen Resolution: 1920 X 1080
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
TopNav: Internet Connection Successful
Local Version: 500, Latest Version: 500
GUI is up to date!
OpenBCI_GUI::Setup: Sample Data exists in Documents folder.
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
The file "/home/martin/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
null
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: /dev/ttyUSB0
ControlPanel: Attempting to Auto-Connect to Cyton
ControlPanel: Found Cyton Dongle on COM port: /dev/ttyUSB0
Cyton AutoConnect Button: get_channel
[DEFAULT]: Attempting to open Serial/COM port: /dev/ttyUSB0
Radios_Config: connect_to_portName: Attempting to open serial port: /dev/ttyUSB0
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Failure: Host on Channel number: 20
[ERROR]: Failed to connect using /dev/ttyUSB0. Check hardware or try pressing 'Autoscan'.
[WARN]: Found a Cyton dongle, but could not connect to the board.
serialList: click! 
[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0
ControlPanel: Found Cyton Dongle on COM port: /dev/ttyUSB0
serialList: click! 
[DEFAULT]: OpenBCI Port Name = /dev/ttyUSB0
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

**java.lang.UnsatisfiedLinkError: liblog.so: cannot open shared object file: No such file or directory
        at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:428)
        at java.base/java.lang.Thread.run(Thread.java:834)
UnsatisfiedLinkError: liblog.so: cannot open shared object file: No such file or directory
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
UnsatisfiedLinkError: liblog.so: cannot open shared object file: No such file or directory**

retiutut, 2020-07-29:
- `No library found for processing.awt.PSurfaceAWT` is ok. We are aware of this. It's used for popup notification window, yet it still works.

- `java.lang.UnsatisfiedLinkError: liblog.so: cannot open shared object file: No such file or directory` is a new issue I have not seen. Maybe it's related to your particular distribution of Linux. 

>I copied all the libraries from OpenBCI_GUI (latest development version) over to the ~/sketchbook/libraries directory for Processing 4 alpha1.

We are going to release GUI v5 using Processing 3, for now, since I'm waiting for a release that includes a fix for https://github.com/OpenBCI/OpenBCI_GUI/issues/674. 

**Can you try a v5.0.0-beta release and/or downgrade to Processing 3?** Maybe this will help.

jacobgmartin, 2020-07-29:
Thanks for your help!

- I tried v5.0.0-beta standalone and there was no problem. 
- I'd still like to get Processing going so I can help with coding on the development version.
- I downgraded Processing to 3 and still had the problem.
- I found the arduino-ndk package on gentoo which provides liblog.so.  I copied this file to the processing directory.  I then ran 
export LD_LIBRARY_PATH="."
./processing

The new error was:

java.lang.UnsatisfiedLinkError: /usr/lib64/libm.so: invalid ELF header
        at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:408)
        at java.lang.Thread.run(Thread.java:748)
UnsatisfiedLinkError: /usr/lib64/libm.so: invalid ELF header
UnsatisfiedLinkError: /usr/lib64/libm.so: invalid ELF header
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.

I tried 32-bit and 64-bit.  

Questions:

- Does anyone know exactly what version of liblog I'm supposed to be using?  I used android-28.  I'm not even sure if this is the right package.

Thanks again!

retiutut, 2020-07-29:
@jacobgmartin It looks like @daniellasry and @Andrey1994 have just merged some code that may resolve this issue for you.

@daniellasry Does @jacobgmartin  just need to try the `upstream/development` branch to test?

daniellasry, 2020-07-29:
Hey @jacobgmartin , Andrey has pushed a fix to the development branch. Let us know if it works for you. Thanks

Andrey1994, 2020-07-29:
but keep in mind that it's not in GUI's code. Its a in library, so you need to copypaste new libs to Processing folder

retiutut, 2020-08-09:
@jacobgmartin https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v5.0.0

jacobgmartin, 2020-08-13:
Works for me now in processing 3.5.3.

Thanks!

closed 2020-08-13 by jacobgmartin