OpenBCI/OpenBCI_GUI #869: OpenBCI GUI unable to start session after connecting

deighv, 2020-09-03:
## Problem

OpenBCI GUI unable to start session after connecting.  Dongle connects to cyton without problem, but once I click start session the application hangs and will not proceed:

[image]

I have installed the FTDI drivers.

## Expected 

The session to begin and for me to start streaming+viewing data.

## Operating System and Version

Windows - 5.0.0

## Running standalone app

Downloaded App (just installed newest version, was able to connect with 4.2)

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

```
Console Log Started at Local Time: 2020-09-03_15-05-17
Screen Resolution: 2560 X 1440
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
TopNav: Internet Connection Not Available
Local GUI Version: 500
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 7
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 8
sourceList: click! 
Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: COM3
Channel count set to 16
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM3
serialList: click! 
[DEFAULT]: OpenBCI Port Name = COM3
Radios_Config: scan_channels
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: set_ovr_channel
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Error reading from Serial/COM port
Radios_Config: system_status
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Error reading from Serial/COM port
Radios_Config: set_ovr_channel
......[2-18]......
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Success: Host override - Channel number: 20
Radios_Config: system_status
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Success: System is Up
[SUCCESS]: Successfully connected to Cyton using COM3
Radios_Config: get_channel
[DEFAULT]: Attempting to open Serial/COM port: COM3
Radios_Config: connect_to_portName: Attempting to open serial port: COM3
Radios_Config: connect_to_portName: Port is open!
Radios_Config: Success: Host and Device on Channel number: 20
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

file: neurosdk-x64.dll is not found in jar file
file: BoardController.dll is not found in jar file
file: GanglionLib.dll is not found in jar file
java.lang.UnsatisfiedLinkError: Unable to load library 'BoardController.dll':
The specified module could not be found.

The specified module could not be found.

The specified module could not be found.

Native library (win32-x86-64/BoardController.dll) not found in resource path ([file:/C:/Program%20Files%20(x86)/OpenBCI/lib/LSLLink.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/OpenBCI_GUI.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/core.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jogl-all.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/gluegen-rt.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jogl-all-natives-windows-amd64.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/gluegen-rt-natives-windows-amd64.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jl1.0.1.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jsminim.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/minim.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/mp3spi1.9.5.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/tritonus_aos.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/tritonus_share.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jssc.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/serial.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/net.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/grafica.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/GifAnimation.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/oscP5.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/udp.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jna-platform.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jna.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/LSLLink.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/jSerialComm.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/brainflow.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/controlP5.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/openbci_gui_helpers.jar, file:/C:/Program%20Files%20(x86)/OpenBCI/lib/ssdp_client.jar])
	at processing.opengl.PSurfaceJOGL$2.run(PSurfaceJOGL.java:408)
	at java.lang.Thread.run(Thread.java:748)
ConsoleLog: Copying console log to clipboard!
ConsoleLog: Copying console log to clipboard!

```

daniellasry, 2020-09-05:
@deighv are you running on 32-bit windows? (we only support 64 bit)

I see you have the gui in `C:/Program Files (x86)/OpenBCI`

Try moving it to `C:/Program Files/OpenBCI_GUI` and see if it makes a difference.

closed 2020-09-25 by retiutut

Ardillen66, 2020-11-12:
Hello, I see the issue is closed, but I had the same problem.

The issue was resolved by running as administrator.

I think the problem for me was that I have an encrypted hard drive and admin privileges are necessary to install and run the GUI.