OpenBCI/OpenBCI_GUI #880: GUI: LSL Sends Corrupt Data When Filters On

coder1923, 2020-09-16:
## Problem

When Filters are toggled on during LSL Networking, the data sent out is corrupt.
This is only noticeable when using Cyton, and NOT when using Synthetic data.

Explain the problem

## Expected 

Expect that data sent over LSL is exactly as what is visualized on the GUI, when Filters are toggled ON.

## Operating System and Version

Windows

## GUI Version

5.0.0

## Running standalone app

Running the downloaded App.

## Type of OpenBCI Board

Cyton.

## Are you using a WiFi Shield?

No, using dongle.

retiutut, 2020-09-19:
There is no supporting data for this claim. Please provide evidence.

coder1923, 2020-09-20:
@retiutut Please check this video I recorded.

https://youtu.be/cL470OmWCNg

As you can see the data on the receiving end is not the same as the data as displayed in the GUI.
Furthermore, even after applying Notch filters to the data on the receiving end, it is different from the one displayed in the GUI.

If I'm not mistaken, the "Filters On" toggle in the GUI is supposed to send data as is displayed on GUI after filters applied, correct?

This is the issue. You can reproduce it by turning on the filters toggle and sending data over LSL. The data being sent, as I'm observing is not filtered.

coder1923, 2020-09-21:
@retiutut I think the issue stems when either the 50Hz toggle is on... or when the Filter toggle is On.
I can't try it on 60Hz in the country I'm in.

retiutut, 2020-10-07:
I was able to use an LSL stream viewer on Windows to see that what you are saying is entirely valid. A screenshot could have helped see the "corrupt data".

After going over the entire data pipeline to investigate LSL on the whole, I discovered a simple coding error that must have come about when we refactored the GUI for v5 from v4.

A fix is about to merge to the development branch. https://github.com/OpenBCI/OpenBCI_GUI/pull/879

Thank you for mentioning that something felt off about LSL streaming, which led to the investigation and fix.

retiutut, 2020-10-07:
Merged!

PR #879 (2020-10-07): Hsc update - Send button + dropdowns
Update Hardware Settings Controller:
- [x] Send button
- [x] Swap out old buttons for cp5 dropdowns
- [x] Lock dropdowns when channel is off
- [x] **Apply Daniel's suggestion https://github.com/OpenBCI/OpenBCI_GUI/pull/879#issuecomment-697994494**
- [x] Short channels automatically when turning them off (like before with buttons)
- [x] User has unchanged settings pop-up  https://github.com/OpenBCI/OpenBCI_GUI/pull/879#issuecomment-704356756
- [x] Require users to Send or Revert hardware settings before closing controller
______

- [x] Fix LSL TimeSeries Fil [... cut]
closed 2020-10-07 by retiutut