OpenBCI/OpenBCI_GUI #889: Synthetic stream problem

rakeshbci, 2020-09-29:
## Problem

I am using the synthetic stream in the GUI to prototype my projects. But when I stream the time series data from any GUI above v 4. 2, the first channel outputs a value from 1-255. This is not a valid time series value.

## Expected 

I would expect the channel one to output the simulated time series data like the rest of the channels.

## Operating System and Version

Windows

## GUI Version

V 5.0.1

## Running standalone app

Standalone app

## Type of OpenBCI Board

Synthetic

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here
[image]

retiutut, 2020-09-29:
@rakeshbci What Networking Mode are you using?

Rakesh420, 2020-09-29:
Lab streaming layer.

retiutut, 2020-09-29:
You might be pulling the Sample Counter instead of the first channel.

retiutut, 2020-09-29:
```
"""Example program to show how to read a multi-channel time series from LSL."""
import time
from pylsl import StreamInlet, resolve_stream

# first resolve an EEG stream on the lab network
print("looking for an EEG stream...")
streams = resolve_stream('type', 'EEG')

# create a new inlet to read from the stream
inlet = StreamInlet(streams[0])

def testLSLSamplingRate():
    start = time.time()
    numSamples = 0
    while time.time() < start + 5:
    # get a new sample (you can also omit the timestamp part if you're not
    # interested in it)
        sample, timestamp = inlet.pull_chunk()
        
        if timestamp:
            print(sample)
            numSamples += 1
    print( numSamples / 5 )

testLSLSamplingRate()
```

```
[[194.0, -9.651573181152344, 30.72825813293457, -32.651493072509766, -1.5084859132766724, 61.2548942565918, -80.80611419677734, 77.72261810302734], [195.0, -10.923236846923828, 26.420150756835938, -16.312349319458008, -22.805356979370117, 55.74642562866211, -60.01969909667969, -19.453380584716797], [196.0, -12.048192977905273, 27.222774505615234, -1.0134592056274414, -41.93016815185547, 51.36628723144531, -0.09689315408468246, -83.00691986083984], [197.0, -12.805867195129395, 20.10445213317871, 16.594785690307617, -56.36474609375, 22.37745475769043, 69.55449676513672, -60.324554443359375], [198.0, -13.800610542297363, 15.998924255371094, 28.138277053833008, -53.751033782958984, -12.688617706298828, 80.85063934326172, -21.704805374145508], [199.0, -13.758541107177734, 9.269229888916016, 36.07243728637695, -35.6177978515625, -43.29778289794922, 77.12794494628906, 44.37217712402344], [200.0, -14.291906356811523, 1.953574776649475, 48.360469818115234, -6.116121292114258, -74.6673355102539, 9.790726661682129, 101.25009155273438], [201.0, -14.168598175048828, -4.5503082275390625, 41.827877044677734, 21.06725311279297, -71.32425689697266, -57.97406005859375, 57.84030532836914], [202.0, -13.8095121383667, -10.735392570495605, 36.85890197753906, 43.71504211425781, -28.57262420654297, -88.84004974365234, -4.434676170349121], [203.0, -12.935434341430664, -18.70047950744629, 24.536582946777344, 46.47328186035156, 13.032248497009277, -69.14616394042969, -94.52349090576172]]
```

retiutut, 2020-09-29:
We need to update the Data Output Guide to reflect this information, also
[image]

Rakesh420, 2020-10-02:
> Lab streaming layer.

Sorry I meant OSC. The LSL works correctly.

retiutut, 2020-11-07:
https://github.com/OpenBCI/OpenBCI_GUI/issues/889#issuecomment-700850806

This is not right for LSL, as there are only 7 channels of data... rather than change the documentation. I will try to exclude the packet number for now.

retiutut, 2020-11-07:
The above fix should work for Unfiltered TimeSeries data for all networking output modes.

PR #911 (2020-11-07): Fix TimeSeries Unfiltered Networking Output Fixes #891 Fixes #889
Fixes #891 

Fixes #889
commit 4ae0176a (2020-11-07): Fix TimeSeries Unfiltered Networking Output Fixes #891 Fixes #889
commit bf86b826 (2020-11-07): Merge pull request #911 from retiutut/development

Fix TimeSeries Unfiltered Networking Output Fixes #891 Fixes #889
closed 2020-11-07 by retiutut