OpenBCI/OpenBCI_GUI #900: Cursor alignment on Mac

espinoj, 2020-10-21:
## Problem

Cursor is pointing above the area of where it is rendered

## Expected 

Rendered cursor and active area are in alightment

## Operating System and Version

macOS

## GUI Version

5.0.1

## Running standalone app

Both processing and downloaded build

## Type of OpenBCI Board

None. Loading files.

## Are you using a WiFi Shield?

No

## Console Log

No errors in log

retiutut, 2020-10-22:
Please provide screenshots or link to a video recording. Not happening on many Macs.

espinoj, 2020-10-23:
I figured out the conditions under which this occurs.  On low resolution screens, the upper dark blue menu bar is not rendered (maybe under the window bar) and the cursor is off calibration in the vertical direction.  The workaround is to increase screen resolution.

retiutut, 2020-10-23:
Still no console log to confirm details....

espinoj, 2020-10-23:
Please see the copied log:

```
Console Log Started at Local Time: 2020-10-23_11-36-31
Screen Resolution: 1440 X 900
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
TopNav: Internet Connection Successful
Local Version: v5.0.1, Latest Version: v5.0.1
GUI is up to date!
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
CP: Playback History Size = 1
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Channel count set to 8
InitSettings: Saving Default Settings to file!
SessionSettings: 3 active widgets saved!
[SUCCESS]: Session started!
[DEFAULT]: Data stream started.
OpenBCI_GUI: openNewLogFile: opened ODF output file: /Users/jue/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2020-10-23_11-36-40/OpenBCI-RAW-2020-10-23_11-36-52.txt
Brainflow start streaming
[DEFAULT]: Data stream stopped.
Brainflow stop streaming
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2020-10-23:
>On low resolution screens,

But the GUI has not been resized and in this example has loaded with the default size. 

```
//set window size
int win_x = 1024;  //window width
int win_y = 768; //window height
```

```
    // If 1366x768, set GUI to 976x549 to fix #378 regarding some laptop resolutions
    // Later changed to 976x742 so users can access full control panel
    if (displayWidth == 1366 && displayHeight == 768) {
        win_x = 976;
        win_y = 742;
    }
    size(win_x, win_y, P2D);
```

espinoj, 2020-10-23:
I have my screen resolution set to 1280x800 and yes I have not resized the screen when the dark blue area is inaccessible.

[image]

retiutut, 2020-10-23:
>Operating System and Version

> macOS

But you leave out Mac OS X version? This is why there is an issue form.

retiutut, 2020-10-23:
>Screen Resolution: 1440 X 900
>I have my screen resolution set to 1280x800 and yes I have not resized the screen when the dark blue area is inaccessible.

? Gui says otherwise.

espinoj, 2020-10-23:
Mac 10.13.6 High Sierra
MacBook Pro (13-inch, 2017, Four Thunderbolt 3 Ports)
3.5 GHz Intel Core i7
16 GB 2133 MHz LPDDR3

retiutut, 2020-10-23:
Try letting this Mac run on the default screen resolution.

retiutut, 2020-10-23:
[image]

Also, you really should update your Mac OS. Your computer is 4 years newer than mine, but 2 major versions behind.

retiutut, 2020-10-29:
Replicated on MacBook Air by changing screen resolution to 1280x800

retiutut, 2020-10-29:
@espinoj Thank you for providing all necessary details to help determine the root cause of this issue. It looks like when I remove the minimum GUI size constraints, it works just fine. So, this can be fixed soon and I have upgraded this to high-priority bug fix. Development branch is currently on v5.0.2-alpha.4, and I would like to resolve this issue and merge a fix in the next week or so.

hack-r, 2020-11-02:
I believe I have the same issue on Windows 10. The mouse pointer has to be significantly below whatever option it's trying to select.

Could have more to do with screen resolution than the OS.

retiutut, 2020-11-06:
The GUI will no longer impose minimum dimensions of the app, but will still wait every ~500ms to resize the drawing area to hopefully avoid crashes.

This seems to be the best fix for everyone, and it somewhat mirrors the behavior of Xcode. Hopefully, this will allow users on lower resolution screens to still fit the GUI when using half of the width of the display.

PR #909 (2020-11-06): 5.0.2 ui bug fixes
- [x] Fix #900 - Drawing GUI on Lower Resolution Screens
- [x] Fix Bug: Update HW Settings objects when loading HW settings
commit e5327da5 (2020-11-06): Remove imposeMinimumGuiDimensions Fixes #900
closed 2020-11-06 by retiutut