OpenBCI/OpenBCI_GUI #904: BDF output format incorrectly encodes 24h time format

rivasd, 2020-11-02:
## Problem

In v5 GUI, when choosing BDF output, the files have a 12h error on the recording start date (e.g. says 7h instead of 19h). This is visible both in latest MNE-Python and EDFBrowser

## Expected 

The current recording start time would be written in the .bdf file

## Operating System and Version

Windows 10 

## GUI Version

5.0.1

## Running standalone app

Standalone

## Type of OpenBCI Board

Cyton+Daisy 16 channels

## Are you using a WiFi Shield?

No

## Console Log

N/A

[image]
 You can see here that the file name has the current start time (around 7PM) but the time reported by EDFBrowser is 7AM

retiutut, 2020-11-05:
@rivasd Thank you for being a valued Community member. I will look into this. It looks like you have included enough information to replicate this issue.

Marking this as high priority and possibly easy fix.

retiutut, 2020-11-06:
https://stackoverflow.com/a/21565579

retiutut, 2020-11-06:
[image]

retiutut, 2020-11-07:
The above fix will be included in 5.0.2. Thank you for creating this issue!

commit 212db202 (2020-11-07): Merge pull request #910 from retiutut/development

Fix Bug: BDF time not in 24 hr format Fixes #904
commit 93397cc1 (2020-11-07): Fix Bug: BDF time not in 24 hr format Fixes #904
closed 2020-11-07 by retiutut