OpenBCI/OpenBCI_GUI #906: UDP of networking widget doesn’t work with playback mode in OpenBCI GUI v5 and v5.0.1.

dynabrain, 2020-11-04:
## Problem

When I use UDP protocol using LabVIEW during Playback mode in GUI v5 and GUI v5.0.1, nothing seems to be streamed through UDP. There is no such an issue with GUI v3 and v4, so I think this is a bug for GUI v5.  
I did not try other protocols such as Serial or LSL with GUI v5, so other protocols may also have the same issue.   

## Expected 

Streaming data during playback mode is very useful to simulate data acquisition while developing an application, so I hope this will be fixed in the next version. 

## Operating System and Version

Windows10

## GUI Version

OpenBCI GUI v5 and v5.0.1

## Running standalone app

I am running the downloaded app. 

## Type of OpenBCI Board

I have a Cyton and a Ganglion board, but this issue only occurs during the playback mode with data files (e.g., OpenBCI_GUI-v5-blinks-jawClench-alpha.txt).

## Are you using a WiFi Shield?

No

## Console Log

```
Console Log Started at Local Time: 2020-11-04_17-51-22
Screen Resolution: 1366 X 768
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: Successfully deleted old GUI v4 sample data files!
OpenBCI_GUI::Setup: Copying sample data to Documents/OpenBCI_GUI/Sample_Data
TopNav: Internet Connection Successful
Local Version: v5.0.1, Latest Version: v5.0.1
GUI is up to date!
ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
[DEFAULT]: Select a file for playback
DataLogging: playbackSelected: User selected C:\Users\isaos\Documents\OpenBCI_GUI\Sample_Data\OpenBCI_GUI-v5-blinks-jawClench-alpha.txt
[SUCCESS]: You have selected "OpenBCI_GUI-v5-blinks-jawClench-alpha.txt" for playback.
Found user playback history file!
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Channel count set to 8
Channel count set to 8
InitSettings: Saving Default Settings to file!
SessionSettings: 4 active widgets saved!
[SUCCESS]: Session started!
OpenBCI_GUI: mouseReleased: screen has been resized...
WidgetManager.pde: Remapping widgets to container layout...
New widget [4] selected for container...
Your cursor is in Playback History
Networking: Protocol mode set to UDP. Stopping network
Stream update numChan to 8
nPointsPerUpdate 10
dataToSend len: 80
[DEFAULT]: Network Stream Started
Networking: StringList size=4 [ "TimeSeries", "127.0.0.1", "12345", "false" ]
[DEFAULT]: UDP successfully connected
[DEFAULT]: Data stream started.
[DEFAULT]: Data stream stopped.
[DEFAULT]: Data stream started.
[DEFAULT]: Data stream stopped.
OpenBCI_GUI: mousePressed: clicked in CP box
OpenBCI_GUI: mousePressed: outside of CP clicked
ConsoleLog: Copying console log to clipboard!
[DEFAULT]: Data stream started.
[DEFAULT]: Data stream stopped.
ConsoleLog: Console closed!
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2020-11-06:
Replicated using the GUI's networking test kit UDP test.

retiutut, 2020-11-07:
Above fix should work!

dynabrain, 2020-11-07:
Awesome!

PR #912 (2020-11-07): Fix getFrameData() for Playback/SD Board Classes Fixes #906
getFrameData() was returning empty data for GUI-made recording playback and Cyton SD card playback... now it returns valid data and Fixes #906
commit 0521b12e (2020-11-07): Merge pull request #912 from retiutut/development

Fix getFrameData() for Playback/SD Board Classes Fixes #906
commit d62582ca (2020-11-07): Fix getFrameData() for Playback/SD Board Classes Fixes #906
closed 2020-12-07 by retiutut