OpenBCI/OpenBCI_GUI #916: Linux Binary - Handle Paths with Spaces

stellarpower, 2020-11-10:
## Problem
The Linux build includes the shell script OpenBCI_GUI inside the zipped file. This needs a trivial modification to support extracting into a location with spaces in the path, but I don't see this file in the repository. It looks like it's being created by some dependency in the build process, so I'm just gonna submit a bug rather than try to work it out when someone else will know already what needs changing.

In the script:
```
#!/bin/sh

APPDIR=$(readlink -f "$0")
APPDIR=$(dirname "$APPDIR")
$APPDIR/java/bin/java -Djna.nosys=true -Djava.ext.dirs="$APPDIR/java/lib/ext"  .....  OpenBCI_GUI "$@"
```
simply needs to become

```
#!/bin/sh

APPDIR=$(readlink -f "$0")
APPDIR=$(dirname "$APPDIR")
"$APPDIR/java/bin/java" -Djna.nosys=true -Djava.ext.dirs="$APPDIR/java/lib/ext"  .....  OpenBCI_GUI "$@"
```
Seems to me all other appearances of $APPDIR are correctly quoted.

## Operating System and Version
Linux: Ubuntu Focal AMD64

## GUI Version
v 5.0.1

## Running standalone app
Yes, the standalone binaries.

retiutut, 2020-11-11:
@stellarpower Where does this script live that you mention? Can you submit a PR if you already have a fix?

retiutut, 2020-11-11:
I can see this in the Linux Standalone. Though, I think this is handled by Processing.

stellarpower, 2020-11-11:
That's the thing, I didn't see it. It's in the zip folder for the release, and I'm presuing it's created by `release_script/make-release.py`, unless I'm just not looking hard enough. It's the same script as in #403.

retiutut, 2020-11-11:
>That's the thing, I didn't see it. It's in the zip folder for the release, and I'm presuing it's created by release_script/make-release.py, unless I'm just not looking hard enough. It's the same script as in #403.

Precisely. I think this means it needs to be a hard-coded fix in make-release.py to edit this file after it is created.

retiutut, 2021-01-22:
>Maybe an easy fix is to do regex in make-release.py for $APPDIR/java/bin/java -> "$APPDIR/java/bin/java" .

For clarification, this regex replace would need to happen in `make-release.py` for Linux build only, after being built.

retiutut, 2021-03-15:
Fixed and Tested!

closed 2021-03-15 by retiutut

stellarpower, 2021-03-19:
Thank you! Sorry been absent, not had much time available for a while!

retiutut, 2021-03-19:
@stellarpower You can see the fix above. As tested, it fully resolves this issue.

Take Care,
RW

commit 48f68e89 (2021-03-15): Fix #916 Enable Launch GUI on Linux from directory with spaces in name
commit cfdc7162 (2021-03-15): update changelog Fix #916