OpenBCI/OpenBCI_GUI #935: mouse location error when maximize the main windows of openbci_gui

hosseinaghaie, 2020-12-13:
## Problem
mouse location error when maximize the main windows of openbci_gui

Explain the problem
when i maximize the openbci_gui ,the location of mouse over is upper than real mouse location.in fact in the maximum  size windows, when i want to select CYTON(LIVE) in DATASOURCE,we must move mouse on GANGLION(LIVE)
[image]

## Expected 
In a perfect world, what do you expect to happen.

## Operating System and Version
windows 7,10

## GUI Version
5.0.1
## Running standalone app
yes
Are you running the downloaded app or are you running from Processing or Visual Studio Code?
processing 
## Type of OpenBCI Board
Cyton/Cyton+Daisy
## Are you using a WiFi Shield?
Yes

## Console Log

Paste any relevant text from the console window here

retiutut, 2020-12-14:
https://github.com/OpenBCI/OpenBCI_GUI/issues/900

closed 2020-12-14 by retiutut

retiutut, 2020-12-14:
https://github.com/OpenBCI/OpenBCI_GUI/pull/909