OpenBCI/OpenBCI_GUI #939: Most of the channels are drained

michal-demecki, 2020-12-18:
## Problem

OpenBCI GUI app does not sync data properly from my MARK IV headset.
Most of the channels are drained the whole session, but some of them randomly get some signals and go drained again after few seconds. 
Is there any way to troubleshoot this issue?
 
It's a brand new headset. Should I flush new software to the Cyton board to make it work with the newest GUI version? 

## Expected 

All channels not drained.

## Operating System and Version

macOS 11.1 (20C69)

## GUI Version

Version 4 (5.0.2)

## Running standalone app

Downloaded app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Session

[OpenBCI-RAW-2020-12-18_23-57-51.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/5718461/OpenBCI-RAW-2020-12-18_23-57-51.txt)

retiutut, 2020-12-21:
>It's a brand new headset. Should I flush new software to the Cyton board to make it work with the newest GUI version?

Do not upload new firmware as the first step to troubleshooting! Our docs specifically state not to do this.

retiutut, 2020-12-21:
@michal-demecki This belongs in openbci.com/forum. I would try asking for help there, and myself and William can help there.

This is likely caused by not connecting earclip electrodes properly.

Closing this issue here, as this is not an issue with the GUI.

closed 2020-12-21 by retiutut