OpenBCI/OpenBCI_GUI #940: Allow Copy/Paste in session name field

BaggioMarco, 2020-12-29:
Dear,

at the moment the session name field doesn't allow copy/paste. Could you please add this feature? You know, many times the participant ID is quite complex and now it has to be written entirely, while it could be copied from other sources and then pasted inside the GUI. Thank you!

retiutut, 2021-01-22:
Not entirely sure this is possible. Involves ControlP5 textfield object. Copy/Pasting in Java is doable:

https://forum.processing.org/one/topic/copy-and-paste.html#25080000001711515.html

retiutut, 2021-03-15:
https://github.com/OpenBCI/OpenBCI_GUI/tree/copy-past-textfield-test

retiutut, 2021-03-16:
Seems possible!!!!

Need to differentiate for OS.

- Mac == Command + V
- Linux == Control + Shift + V
- Windows == Control + V

BaggioMarco, 2021-03-16:
Hi @retiutut that's a great news! 
If I update now from the master branch will I get this feature, or I've to download this one here https://github.com/OpenBCI/OpenBCI_GUI/tree/copy-past-textfield-test ???
In case, when this feature will be available in the master branch? thank you

retiutut, 2021-03-16:
@BaggioMarco https://github.com/OpenBCI/OpenBCI_GUI/milestone/8

You will see the commits here that directly resolve this issue.

You're welcome! 😄 👍

BaggioMarco, 2021-03-16:
Awesome! Thank you!

retiutut, 2021-03-17:
[image]

retiutut, 2021-03-17:
[image]

PR #960 (2021-03-19): Implement Copy-Paste for all Textfields
Fixes #940, but also applies the fix to all textfields!

To Do:
- [x] Implement code that accounts for Windows, Linux, and Mac
- [x] Make sure no garbage characters are included when copying or pasting
- [x] Test Copy and Paste on Windows
- [x] Test Copy and Paste on Linux
- [x] Test Copy and Paste on Mac

____
Extra Credit:
- [x] Avoid playback history file not found exception Fixes #959
closed 2021-03-19 by retiutut