OpenBCI/OpenBCI_GUI #941: Expert Mode : unplanned keys crash GUI

pirelaurent, 2021-01-15:
## Problem

Using expert mode with a French keyboard, trying to identify equivalent places, i crash GUI while pressing unplanned letters.

## Expected 

No effect on wrong key 

## Operating System and Version

Win10 Pro

## GUI Version

Local Version: v5.0.1,

## Running standalone app

Dowloaded app  .Exe 

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Interactivity: 'Z' Pressed...sending to Board...
Sending config string to board: Z
java.lang.ArrayIndexOutOfBoundsException: 10
	at OpenBCI_GUI$ADS1299Settings.setChannelActive(OpenBCI_GUI.java:1320)
	at OpenBCI_GUI$BoardCyton.setEXGChannelActive(OpenBCI_GUI.java:2753)
	at OpenBCI_GUI.parseKey(OpenBCI_GUI.java:10596)
	at OpenBCI_GUI.keyPressed(OpenBCI_GUI.java:10484)
	at processing.core.PApplet.keyPressed(PApplet.java:3121)
	at processing.core.PApplet.handleKeyEvent(PApplet.java:2996)
	at processing.core.PApplet.dequeueEvents(PApplet.java:2648)
	at processing.core.PApplet.handleDraw(PApplet.java:2486)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

retiutut, 2021-01-15:
@pirelaurent Please provide the full console log.

pirelaurent, 2021-01-15:
hi retiutut : 
here the console log V5.0.1
[Console_2021-01-15_22-54-07.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/5823118/Console_2021-01-15_22-54-07.txt)
Same thing after installing V5.0.2 : 
[Console_2021-01-15_23-05-42.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/5823144/Console_2021-01-15_23-05-42.txt)
HTH

retiutut, 2021-01-15:
@pirelaurent What French characters/keys have you pressed?

Here is a visual comparison of French vs English keyboard. Maybe this will help: http://www.farah.cl/Keyboardery/A-Visual-Comparison-of-Different-National-Layouts/#frFR

retiutut, 2021-01-15:
>Virtual key codes do not identify a physical key: they depend on the platform and keyboard layout. For example, the key that generates VK_Q when using a U.S. keyboard layout will generate VK_A when using a French keyboard layout.

https://docs.oracle.com/javase/8/docs/api/index.html?java/awt/event/KeyEvent.html

pirelaurent, 2021-01-18:
The keyboard doesn't matters for me, i'm able to find where an english key
is located on my keyboard, but in case of unknown key by OpenBCI, just
expect that this don't crash the app.
Maybe you must add some control list ?

retiutut, 2021-01-18:
@pirelaurent What French characters/keys have you pressed that caused the GUI to crash? 

In the console logs you provided, I saw `8` and `Z` which should not have caused the GUI to crash.

pirelaurent, 2021-01-18:
I redo the same with 5.0.2 : start expert mode, strike keys 7 then 8 : ok
that disable both channels,
strike *q* on my french keyboard (ie z on english )  and crash : I try
lowecase and uppercase : the same.

GUI Settings: Expert Mode On
[DEFAULT]: Expert Mode ON: All keyboard shortcuts and features are enabled!
Sending config string to board: x7161000X
WARNING: Lost 135 Samples Between 120-0
Sending config string to board: x8161000X
WARNING: Lost 212 Samples Between 44-1
java.lang.ArrayIndexOutOfBoundsException: 8
at OpenBCI_GUI$ADS1299Settings.setChannelActive(OpenBCI_GUI.java:1311)
at OpenBCI_GUI$BoardCyton.setEXGChannelActive(OpenBCI_GUI.java:2976)
at OpenBCI_GUI.parseKey(OpenBCI_GUI.java:11110)
at OpenBCI_GUI.keyPressed(OpenBCI_GUI.java:11054)
at processing.core.PApplet.keyPressed(PApplet.java:3121)
at processing.core.PApplet.handleKeyEvent(PApplet.java:2996)
at processing.core.PApplet.dequeueEvents(PApplet.java:2648)
at processing.core.PApplet.handleDraw(PApplet.java:2486)
at
processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
at java.util.TimerThread.mainLoop(Timer.java:555)
at java.util.TimerThread.run(Timer.java:505)
****************
(This occurs with a cython on line, not with a playback files as all
keyboards keys seems to be not available )

retiutut, 2021-01-18:
Good news! I think everything is working for the most part with your French keyboard.

The issue you have when pressing `q` is that is the command to set Channel 9 active, and you are only using Cyton 8 Ch.

I can make a fix to ensure this doesn't cause the crash you have shared. This can be included in the next GUI - v5.0.3!

Take Care,
RW

👍🛠

retiutut, 2021-01-18:
I have replicated this issue on Mac with Windows keyboard:

```
[SUCCESS]: Session started!
GUI Settings: Expert Mode On
[DEFAULT]: Expert Mode ON: All keyboard shortcuts and features are enabled!
Interactivity: 'Z' Pressed...sending to Board...
Sending config string to board: Z
Interactivity: 'z' Pressed...sending to Board...
Sending config string to board: z
ADS1299SettingsBoard.pde:236:0:236:0: ArrayIndexOutOfBoundsException: 8
```

commit 9daefb8d (2021-01-22): Fix Bug - Expert mode unplanned keys crash GUI

Fixes #941
closed 2021-01-22 by retiutut