OpenBCI/OpenBCI_GUI #942: Several errors in Settings through load/save/default

pirelaurent, 2021-01-15:
## Problem
@see pdf 
[bugSetting.pdf](https://github.com/OpenBCI/OpenBCI_GUI/files/5819943/bugSetting.pdf)
1- default settings when OpenBCI is started 
2- change parameters in upper options, Time series options included active channels, FFT option
3-save settings in the json file ( extension changed below from .json to .txt to be loaded in this github) 
[CytonUserSettings_000.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/5819973/CytonUserSettings_000.txt)
4- restore default from setting menu:  
Upper options are not restored to default (Notch,BO filter, Smoothing (gain not tested ))
In Time series:  vertical scale is restored,BUT Window option is overriden by vertical scale value,  active channels are not restored,  
In FFT : values are restored correctly. 
5- Close then restart openGUI : same as initial 
6- Load the previously setting saved  in step 3 
Upper options are not restored to default (Notch,BO filter, Smoothing (gain not tested ))
In Time series:  vertical scale is restored,BUT Window option is overriden by vertical scale value,  active channels are not restored,  
In FFT : values are restored correctly. 

## Expected 
Must be correctly saved and restored.

## Operating System and Version
Win10 Pro

## GUI Version
V5.0.1

## Running standalone app
Standalone .exe 

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log
SessionSettings: loadConfigFile: User selected C:\Users\plaurent\Documents\OpenBCI_GUI\Settings\CytonUserSettings_000.json
LoadGUISettings: Applied Widget 0 to Container 4
LoadGUISettings: Applied Widget 1 to Container 3
LoadGUISettings: Applied Widget 2 to Container 9
LoadGUISettings: Expert Mode Off
Networking: Protocol mode set to Serial. Stopping network
Apply Serial Networking Mode
[SUCCESS]: Settings Loaded!

retiutut, 2021-01-15:
I remember writing some code that is supposed to delete an incompatible GUI settings file, if detected.

I remember writing some code that is supposed to delete an incompatible GUI settings file, if detected.I will try to see what happens in GUI 5.0.2. This doesn't seem like a mismatch.

retiutut, 2021-01-15:
```
        ////////Apply Time Series widget settings
        w_timeSeries.setTSVertScale(loadTimeSeriesSettings.getInt("Time Series Vert Scale"));
        w_timeSeries.cp5_widget.getController("VertScale_TS").getCaptionLabel().setText(w_timeSeries.getTSVertScale().getString()); //changes front-end
        
        w_timeSeries.setTSHorizScale(loadTimeSeriesSettings.getInt("Time Series Horiz Scale"));
        w_timeSeries.cp5_widget.getController("Duration").getCaptionLabel().setText(w_timeSeries.getTSVertScale().getString());
        ```
        
        There is an error on this last line it should be `getTSHorizScale().getString()`

retiutut, 2021-01-15:
Everything else you have mentioned was purposefully excluded from saving/loading of GUI "Session Settings".

It is possible to re-enable Notch, BP, and Smoothing.

____

RE: Channel On/Off

Please update to GUI 5.0.2 and "Clear All Settings". 

Using GUI 5.0.2, you can Save, Load, and then Send custom hardware commands to the Cyton. This includes channel on/off. 😄 

___

Thank you for making this issue! I hope to have this resolved soon.

Take Care,
RW

pirelaurent, 2021-01-18:
Hi RW, i test a bit more V5.0.2 about settings : 
**Startup :** 
- the values at startup are not exactly the same as the defaults value : if no default file, file will be created with a FFT max freq at 40 (entry 1 in menu) while at startup it is 60 (Entry 0 in menu, maybe not initialised ? ) : could be improved by loading default at startup ( or create it then reload it immediately )? 

**Save and restore settings about channels :** 
 After several tests, i find my misunderstanding of what are active and/or selected channels:  
Docs talk about 
- 'deactivate channels' ( clicking on numbered icone or using keyboard) 
- 'toggle channels on/off' ( drop down menu Channels' 
and **these are not the same** ! 
=> Only channels not toggle off (by Channels menu) are visible and saved/restored in Settings (and it's ok)

I understand that the channels have 3 states one can see in channels drop down menu : 
 - background blue : channel is active and only active channels go into saved settings 
 - background grey : channel is not active (disappear in layout) : don't go into saved settings 
 In any kind of background : 
  border red : channel's values are not displayed and acquired as 0 ( I believe, not tested)
  border blue : channel's values are acquired and displayed (if channel is visible of course).

=> Maybe some enhancement in docs will help others.

retiutut, 2021-01-26:
> the values at startup are not exactly the same as the defaults value : if no default file, file will be created with a FFT max freq at 40 (entry 1 in menu) while at startup it is 60 (Entry 0 in [...]

This was caused by loading settings in from the Spectrogram widget, which would override the setting in FFT. This has been fixed, along with the rest of the bugs that I found investigating this ticket.

Again, thank you for the awesome PDF image. This allowed me to resolve the issues in a timely manner! 😇

```
Save and restore settings about channels :
After several tests, i find my misunderstanding of what are active and/or selected channels:
Docs talk about

'deactivate channels' ( clicking on numbered icone or using keyboard)
'toggle channels on/off' ( drop down menu Channels'
and these are not the same !
=> Only channels not toggle off (by Channels menu) are visible and saved/restored in Settings (and it's ok)
I understand that the channels have 3 states one can see in channels drop down menu :

background blue : channel is active and only active channels go into saved settings
background grey : channel is not active (disappear in layout) : don't go into saved settings
In any kind of background :
border red : channel's values are not displayed and acquired as 0 ( I believe, not tested)
border blue : channel's values are acquired and displayed (if channel is visible of course).
=> Maybe some enhancement in docs will help others.
```

I will try to clarify this in either the GUI main Doc or the GUI widget guide! Thank you for getting this started!

PR #947 (2021-01-26): Session settings fix jan2021
Fixes #942
commit a813659e (2021-01-26): Fix Session Settings Errors found in 5.0.2

Fixes #942
closed 2021-01-29 by retiutut