OpenBCI/OpenBCI_GUI #943: Unexpected error during push_chunk: The number of buffer elements to send is not a multiple of the stream's channel count.

athanoid, 2021-01-15:
## Problem

I have connected the Pulse Sensor and it is working fine in the GUI. I can see the analog signal and the BPM.
However, when I try to send the pulse data via LSL, I get the following error: 

> Unexpected error during push_chunk: The number of buffer elements to send is not a multiple of the stream's channel count.

From what I understand, probably the lsl code is expecting more channels?

## Expected 

Stream Pulse Sensor data via LSL

## Operating System and Version

Ubuntu 20.04 - 64bit

## GUI Version

v5.0.2 - 2020/12/16

## Running standalone app

Running the standalone app.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Unexpected error during push_chunk: The number of buffer elements to send is not a multiple of the stream's channel count.

retiutut, 2021-01-15:
@athanoid Can you please provide the full console log?

retiutut, 2021-01-15:
I was not able to replicate this issue, but I did have issues with the pulse widget itself in 5.0.2. Thank you for making this issue!

-RW

wjcroft, 2021-01-16:
Is the issue in the Pulse Widget, or the Networking Widget?

retiutut, 2021-01-16:
- Pulse Widget in GUI v5.0.2 does not function properly, as seen in GUI v4.2.0.
- I received 0 data using LSL Networking Test Script (found in this repository) from Pulse Sensor data type in Networking, even with the widget opened.
- Output promised in Output Guide is not happening right now

🛠

athanoid, 2021-01-18:
>     * Pulse Widget in GUI v5.0.2 does not function properly, as seen in GUI v4.2.0.
> 
>     * I received 0 data using LSL Networking Test Script (found in this repository) from Pulse Sensor data type in Networking, even with the widget opened.
> 
>     * Output promised in Output Guide for LSL is not happening right now for Pulse Sensor (should be 3 values)
> 
> 
> Maybe I did something wrong, so I will test on other computers also.
> 
> Initially, it looks like the BPM processing is happening too often or not enough, and this jacks up the data that would come from the widget. This is related to NodeJS Drivers -> BrainFlow transition in the GUI.
> hammer_and_wrench

Thanks for looking at it.

Will it be also possible that the 3 channels sent via LSL, namely: raw signal; BPM and IBI are using different sampling frequencies and the LSL stream is instantiated with a fixed one? (125hz)

retiutut, 2021-01-18:
@athanoid It is still important to have the full console log. Can you please provide one?

athanoid, 2021-01-18:
> @athanoid It is still important to have the full GUI console log. Can you please provide one?

oops, here it goes:
```
Console Log Started at Local Time: 2021-01-18_16-18-11
Screen Resolution: 1920 X 1080
Operating System: Linux
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
ControlP5 2.3.1 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
TopNav: Internet Connection Successful
Local Version: v5.0.2, Latest Version: v5.0.2
GUI is up to date!
The file "/root/Documents/OpenBCI_GUI/Settings/UserPlaybackHistory.json" is missing or inaccessible, make sure the URL is valid or that the file has been added to your sketch and is readable.
OpenBCI_GUI::Control Panel: Playback history file not found or other error.
null
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: /dev/ttyUSB0

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: /dev/ttyUSB0
Success: Host and Device on Chan using COM port: /dev/ttyUSB0
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Channel count set to 8
InitSettings: Saving Default Settings to file!
SessionSettings: 3 active widgets saved!
[SUCCESS]: Session started!
New widget [10] selected for container...
Your cursor is in Time Series
Sending config string to board: /2
[DEFAULT]: Starting to read analog inputs on pin marked A5 (D11), A6 (D12) and A7 (D13)
[DEFAULT]: openBCI_GUI: startButton was pressed. Starting data transfer, wait a few seconds.
Brainflow start streaming
[DEFAULT]: Data stream started.
OpenBCI_GUI: openNewLogFile: opened ODF output file: /root/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2021-01-18_16-18-12/OpenBCI-RAW-2021-01-18_16-18-44.txt
New widget [3] selected for container...
Your cursor is in FFT Plot
Networking: Protocol mode set to LSL. Stopping network
Stream update numChan to 8
nPointsPerUpdate 10
dataToSend len: 80
Networking: StringList size=5 [ "Pulse", "obci_eeg1", "PPG", "3", "false" ]
[DEFAULT]: Network Stream Started
[DEFAULT]: Network Stream Stopped
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2021-01-18:
So where did this error come from?

>Unexpected error during push_chunk: The number of buffer elements to send is not a multiple of the stream's channel count.

athanoid, 2021-01-18:
> So where did this error come from?
> 
> > Unexpected error during push_chunk: The number of buffer elements to send is not a multiple of the stream's channel count.

From the linux console:

[image]

retiutut, 2021-01-18:
@athanoid I bet I will get the same error if I try on Linux. 

Thanks so much for providing information in a timely manner. This means I can start investigating a fix ASAP.

😇

retiutut, 2021-01-22:
I was able to replicate this issue on Linux!

PR #946 (2021-01-24): 2021 pulse sensor update - GUI v5.0.3
Make it so that the Pulse Sensor Widget + Networking output for Pulse data type is both functional and usable in GUI v5.

Fixes #943 

Testing:
- [x] OSC using MaxMSP
- [x] UDP using `Networking-Test-Kit/UDP/udp_receive_pulse.py`
- [x] LSL using `Networking-Test-Kit/LSL/lslStreamTest_AnalogPinPulse.py`
~~- [ ] Serial (difficult to test without Virtual Com Port).~~

All tests passing for this data type. LSL seems to drop a few packets. UDP and OSC are close to flawless sampling rate. Tested on Mac. Serial output for Pulse data should work as expected.

Migh [... cut]
closed 2021-01-24 by retiutut