OpenBCI/OpenBCI_GUI #944: Wrong Data format for Accel/aux channel in UDP networking

pirelaurent, 2021-01-18:
## Problem
Wrong sent data for Accel/aux in udp 
Minor : 
Some python code (python 3.5 on my machine) to change (see below) :    
   line 14 16 print(with parenthesis ) instead of simple print
  try to avoid error the _JSON object must be str, not 'bytes'_ by adding  .decode. ( but not tested everywhere) 
Major : 
Data values are sent with a localized decimal separator which is comma in Europe, while data format is CSV, resulting in wrong data with too many delimiters.   
Output ( can see commas everywhere)  
```
b'{"type":"accelerometer","data":[-0,002,0,932,0,302]}\r\n'
Expecting ',' delimiter: line 1 column 37 (char 36)
``` 
Other channels ( fft, eeg, bandpower ) have same dot decimal delimiter like [45383.02, 37020.43, -22483.17, etc. 

## Expected 
Standardized output with same dot delimiters. ( or if localized numbers everywhere, don't use csv but | separated values)

Here patched code for minor pbs. 
``` def print_message(*args):
    try:
        print(args[0]) #added to see raw data 
        obj = json.loads(args[0].decode())
        print(obj.get('data'))
    except BaseException as e:
        print(e)
```

## Operating System and Version

Win10

## GUI Version

v5.0.2

## Running standalone app
standAlone
## Type of OpenBCI Board

Cyton
## Are you using a WiFi Shield?
No

## Console Log
Trace of one full message : 
b'{"type":"accelerometer","data":[0,002,0,930,0,294]}\r\n'
Expecting ',' delimiter: line 1 column 36 (char 35)
b'{"type":"auxiliary","data":[0000,0000,0000]}\r\n'
Expecting ',' delimiter: line 1 column 30 (char 29)
b'{"type":"auxiliary","data":[0,0,0,0,0]}\r\n'
[0, 0, 0, 0, 0]

retiutut, 2021-01-19:
Is this in the `udp_recieve.py` Networking Test Kit?

pirelaurent, 2021-01-20:
Hello Richard, yes, it is using receive. py. as it is the same code for all channels and only the accel/aux is wrong, i believe  the pb is rather  on the send side. ( not critical for me, i'll use osc. Rather to have stronger international code) Envoyé depuis mon téléphone Huawei-------- Message original --------De : Richard Waltman <notifications@github.com>Date : mer. 20 janv. 2021 à 00:57À : OpenBCI/OpenBCI_GUI <OpenBCI_GUI@noreply.github.com>Cc : PIERRE LAURENT <pi.r.laurent@gmail.com>, Author <author@noreply.github.com>Objet : Re: [OpenBCI/OpenBCI_GUI] Wrong Data format for Accel/aux channel in UDP networking   (#944)
Is this in the udp_recieve.py Networking Test Kit?

retiutut, 2021-01-22:
I expect the following code to work:

```
       // UDP
            } else if (this.protocol.equals("UDP")) {
                String outputter = "{\"type\":\"accelerometer\",\"data\":[";
                for (int i = 0; i < NUM_ACCEL_DIMS; i++) {
                    float accelData = w_accelerometer.getLastAccelVal(i);   
                    Locale currentLocale = Locale.getDefault();
                    DecimalFormatSymbols otherSymbols = new DecimalFormatSymbols(currentLocale);
                    otherSymbols.setDecimalSeparator('.');
                    otherSymbols.setGroupingSeparator(','); 
                    DecimalFormat df = new DecimalFormat("0.000", otherSymbols);
                    String accelData_3dec = df.format(accelData);
                    //String accelData_3dec = String.format("%.3f", accelData);
                    outputter += accelData_3dec;
                    if (i != NUM_ACCEL_DIMS - 1) {
                        outputter += ",";
                    } else {
                        outputter += "]}\r\n";
                    }
                }
                try {
                    this.udp.send(outputter, this.ip, this.port);
                } catch (Exception e) {
                    println(e.getMessage());
                }
```

[image]

retiutut, 2021-01-22:
About AnalogRead output over UDP and then received in python:
>You cannot have leading zeros at all in python3 and in python 2 they signify octal numbers. If you are going to pass it into a string just keep it as a string.
https://stackoverflow.com/a/29812014

retiutut, 2021-01-22:
```
        threeDecimalPlaces = new DecimalFormat("0.000", otherSymbols);
        fourLeadingPlaces = new DecimalFormat("####", otherSymbols);
        ```

PR #945 (2021-01-22): 5.0.3 Alpha 2
Update UDP recieve Test and use DecimalFormat in Networking Widget
Fixes #944

- [x] Fix UDP receive Test
- [x] Pass Test
- [x] Update networking output guide
_____ 

- [x] Fix Expert Mode unplanned keyboard shortcuts crash GUI #941

____

- [x] Update changelog and bump version to 5.0.3-alpha.2
commit 2c13f7c4 (2021-01-22): Update UDP recieve and use DecimalFormat in Networking Widget

Fixes #944
closed 2021-01-22 by retiutut