OpenBCI/OpenBCI_GUI #950: Some of GUI Keyboard Shortcuts don't work

Otemo, 2021-02-01:
## Problem

Some of the GUI Keyboard Shortcuts do not work

## Expected 
under Expert Mode setting...
send '[' ']' to enable/disable synthetic square wave
send 'z' 'Z' to start/stop impedance test

## Operating System and Version

Windows10

## GUI Version
v5.0.3

## Running standalone app

Both Downloaded app and Running from Processing

## Type of OpenBCI Board

Ganglion

## Are you using a WiFi Shield?

No

## Console Log
I can find the following message at the bottom of GUI screen.

"Expert Mode: '[' pressed. This is not assigned or applicable to current setup"
"Expert Mode: ']' pressed. This is not assigned or applicable to current setup"

retiutut, 2021-02-01:
Here is the code responsible for this breaking change:

```
    if (currentBoard instanceof Board) {
        output("Expert Mode: '" + key + "' pressed. This is not assigned or applicable to current setup.");
        //((Board)currentBoard).sendCommand(str(key));
    }
    ```
    
    As you can see, this avoids sending any bogus commands to the Cyton. At the same time, there was no case for these commands. I think a case should be added for these explicit commands. 
    
   I will also consult with the OpenBCI team on if the GUI should send unrecognized commands to boards.
   
   @Otemo Thanks for noticing this and making an issue!

Take Care,
RW

retiutut, 2021-02-04:
@Otemo For now, you can use the "Send Custom Command" UI at the bottom of the Hardware Settings view. This is only available when using Cyton and Expert Mode.

Otemo, 2021-02-04:
Hi retiutut,
Thank you very much for the useful information. Looking forward to the completion of version 5.0.4 and I will send the necessary commands in the way you taught!

retiutut, 2021-03-15:
Sorry just now realizing you are using Ganglion. I'm not sure we should be sending the impedance commands when the Ganglion Signal Widget exists.

retiutut, 2021-03-15:
```
public void setCheckingImpedance(boolean checkImpedance) {
        if (checkImpedance) {
            if (isCheckingImpedance) {
                println("Already checking impedance.");
                return;
            }
            if (streaming) {
                stopRunning();
            }
            sendCommand("z");
            startStreaming();
        }
        else {
            if (!isCheckingImpedance) {
                println ("Impedance is not running.");
                return;
            }
            if (streaming) {
                stopStreaming();
            }
            sendCommand("Z");
        }
        isCheckingImpedance = checkImpedance;
    }
    ```

retiutut, 2021-03-15:
Adding this code
```
    if (currentBoard instanceof BoardGanglion) {
        if (val == '[' ||  val == ']'){
            output("Expert Mode: '" + val + "' pressed. Sending to Ganglion...");
            ((Board)currentBoard).sendCommand(str(val));
            return;
        }
    }
    ```

retiutut, 2021-03-15:
Added keyboard shortcuts for square wave for Ganglion board. Fixed!

closed 2021-03-15 by retiutut