OpenBCI/OpenBCI_GUI #951: Unable to run the lslStreamTest_FFTplot.py python script under networking-Test-Kit

SreeHaran, 2021-02-02:
## Problem

When I try to run the OpenBCI_GUI/Networking-Test-Kit/LSL/lslStreamTest_FFTplot.py, it shows an exception. Unable to run that code. It is like a deprecation warning(I have pasted the console output down). Please update that code.. Or say whether I'm doing something wrong while running that code.

## Expected 

A pop-up window showing the graph of FFT using matplotlib

## Operating System and Version

Windows

## GUI Version

openBCI GUI v5.0.1

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

Yes

## Console Log

VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
  return array(a, dtype, copy=False, order=order, subok=True)
TypeError: float() argument must be a string or a number, not 'list'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "fft.py", line 36, in <module>
    plt.plot(channel_data[chan][:60])
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\pyplot.py", line 2840, in plot
    return gca().plot(
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\axes\_axes.py", line 1745, in plot
    self.add_line(line)
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\axes\_base.py", line 1964, in add_line
    self._update_line_limits(line)
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\axes\_base.py", line 1986, in _update_line_limits
    path = line.get_path()
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\lines.py", line 1011, in get_path
    self.recache()
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\lines.py", line 658, in recache
    y = _to_unmasked_float_array(yconv).ravel()
  File "F:\software\Miniconda\envs\bci\lib\site-packages\matplotlib\cbook\__init__.py", line 1289, in _to_unmasked_float_array
    return np.asarray(x, float)
  File "F:\software\Miniconda\envs\bci\lib\site-packages\numpy\core\_asarray.py", line 83, in asarray
    return array(a, dtype, copy=False, order=order)
ValueError: setting an array element with a sequence.

retiutut, 2021-02-03:
I just tried using Python 2 and 3 with no issues. Try again using the files found in the current `master` branch using version 5.0.3.

closed 2021-02-03 by retiutut

SreeHaran, 2021-02-03:
what version of matplotlib, are you using?

retiutut, 2021-02-03:
The latest one?