OpenBCI/OpenBCI_GUI #953: Not able to run the OpenBCI GUI on Processing Ide

rakeshbci, 2021-02-11:
## Problem

I tried to run the OpenBCI_GUI.pde, I followed the guide and still I am getting an error. The function setBorderColor(int) does not exist.

## Expected 

Expect to run the GUI

## Operating System and Version

Windows

## GUI Version

Latest code from github

## Running standalone app

Running the code in Processing Ide

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

retiutut, 2021-02-12:
Hey there @rakeshbci !

I decided to heavily modify the ControlP5 library for GUI usage. You probably need to re-copy the Libraries to USER/Documents/Processing/libraries.

Hope this helps,
RW

rakeshbci, 2021-02-13:
I have copied the libraries.

retiutut, 2021-02-13:
Please provide a Console Log so we can see the error in context.

retiutut, 2021-02-14:
Here is the commit where the method was added to ControlP5 library for the Button class: https://github.com/OpenBCI/OpenBCI_GUI/commit/d0ea37440799e24d46a17e600fc54c17049e488a#diff-57d22759b94d8966052efa81471c76d4046ee36c5c4661523b9c7d0522c653c2

closed 2021-02-14 by retiutut