OpenBCI/OpenBCI_GUI #963: Adding Electrode Position Labels instead of Electrode Numbers in GUI

retiutut, 2021-03-31:
**Is your feature request related to a problem? Please describe.**

Full details can be found here on the forum, along with my initial thoughts and potential hang-ups involved in developing this feature.
https://openbci.com/forum/index.php?p=/discussion/2930/adding-labels-instead-of-electrode-numbers#latest

**Describe the solution you'd like**
Add ability to type in the 10-20 electrode names in TimeSeries or somewhere else. Might even be helpful to save/load this information if it doesn't cause problems elsewhere in the GUI.

**Describe alternatives you've considered**
Currently, GUI uses channel numbers as the "single source of truth" for all users. This works great when users may be trying out new electrode positions for studies and use-cases.

**Additional context**
This would likely help more than a few users, which could justify adding this feature for all users.

retiutut, 2022-08-15:
Relates to https://github.com/OpenBCI/OpenBCI_GUI/issues/648

pgclemens, 2023-11-24:
I would like to see this feature

tkgesis, 2024-06-04:
So would I. 
Possibly, the collection of montages from https://mne.tools/stable/auto_tutorials/intro/40_sensor_locations.html#working-with-built-in-montages can be reused. Ideally, it would go beyond a mapping of channel to electrode labels, and instead 1) take into account scalp positions in visualisations and metrics and 2) the electrode labels would also be included in the LSL stream info description.