OpenBCI/OpenBCI_GUI #966: Discussion: Should we reorganize how data is stored in Documents/OpenBCI_GUI?

retiutut, 2021-04-23:
## Problem

Currently, different types of files are stored in separate folders. This might make it harder to find the console log that correlates to multiple GUI sessions within an instance of the app.

## Expected 

Use something like BIDS https://bids.neuroimaging.io to store data in a predictable and accessible manner. Also, consider input from Community members, partners, and team members.

## Operating System and Version

All OS

## GUI Version

GUI v5.0.4

## Running standalone app

Dev and Standalone

## Type of OpenBCI Board

All Boards

## Are you using a WiFi Shield?

N/A

## Console Log

N/A

retiutut, 2021-04-23:
[image]

Initial proposal that does not consider BIDS. Maybe there is a middle ground between this proposal and the BIDS specification. Also, BIDS would allow us to store more useful information in a sidecar.json file.

markopetrovic-neurotech, 2021-06-10:
How about a BIDS class similar to the SessionSettings class that saves to a folder in session recordings?