OpenBCI/OpenBCI_GUI #968: Ultra-hd uhd hdpi screen

allali, 2021-04-30:
Hi,
Please considere adding the following line in setting() method:

    pixelDensity(displayDensity());

So the app will scale correclty on ultra-hd display (like mine) :)

allali, 2021-04-30:
Also,
Add the: 
    surface.setResizable(true);
into the setup() function...

retiutut, 2021-05-03:
>Also,
Add the:
surface.setResizable(true);
into the setup() function...

This is already done is delayedSetup() as a new Thread.

allali, 2021-05-04:
On my system: ubuntu 18.04.5 LTS with gnome graphical session, the setResizable is not effective in delayedSetup() but works in setup()...

retiutut, 2021-05-05:
@allali Please look at this comment from a former Senior Engineer at OpenBCI. https://github.com/OpenBCI/OpenBCI_GUI/issues/145#issuecomment-487415123

retiutut, 2021-05-07:
@allali It seems you did not follow the Issue template.

Can you please send screenshots of what you see on your screen since this involves graphics?

Also, it would be helpful run the current version of GUI (5.0.4) from Processing and change the code so we can see a "before and after" for the proposed change.

Take Care,
RW

allali, 2021-05-08:
I ran the test using openbci gui master branch and preprocessing 3.5.4

Without any modification, the app is resizable but the resolution is not adapted for uhd screen:
[image]

When I add the density correction, resolution is now correct but the app is not resizable anymore:
[image]

With the two fixes (density and resizeable in setup), everything is correct:
[image]

The fixes suggest in #145 are quite poor: exec param under windows and run_scaled under linux which is really the last solution for uhd issues (slow responsive, huge resources). I don't have windows to check the correctness of the two modifications I suggest but in theory they are the right way to manage uhd regarding preprocessing guideline/API.

retiutut, 2021-06-03:
Maybe someone at the OpenBCI office has an UltraHD monitor so we can double-check the proposed fix before deploying to all users. I have scheduled this to be part of next month's update.

retiutut, 2021-06-21:
@allali What is the resolution of your monitor?

allali, 2021-06-21:
3840x2160

retiutut, 2021-06-30:
Here is the supporting documentation to validate this finding and apply the suggested fix in the original post:

https://processing.org/reference/pixelDensity_.html
https://processing.org/reference/displayDensity_.html

>This function is new with Processing 3.0. It makes it possible for Processing to render using all of the pixels on high resolutions screens like Apple Retina displays and Windows High-DPI displays. This function can only be run once within a program and it must be used right after size() in a program without a setup() and used within setup() when a program has one. The pixelDensity() should only be used with hardcoded numbers (in almost all cases this number will be 2) or in combination with displayDensity() as in the third example above. 

>When the pixel density is set to more than 1, it changes all of the pixel operations including the way get(), set(), blend(), copy(), and updatePixels() all work. See the reference for pixelWidth and pixelHeight for more information. 

>To use variables as the arguments to pixelDensity() function, place the pixelDensity() function within the settings() function. There is more information about this on the settings() reference page.

retiutut, 2021-06-30:
Here is the code the resolve this issue, which will be at the beginning of setup(), just after size() is called in settings().

```
void setup() {
    //Account for high-dpi displays on Mac, Windows, and Linux Machines Fixes #968
    pixelDensity(displayDensity());
    StringBuilder sb_dpi = new StringBuilder("OpenBCI_GUI::Setup: High-DPI Screen == ");
    sb_dpi.append(displayDensity() == 2);
    println(sb_dpi);
 ```

retiutut, 2021-06-30:
This status will be printed on every GUI app start. If a high-dpi screen is detected among all available displays, it will increase pixel density (as stated in Processing codebase https://github.com/processing/processing/blob/ad31c61bc10e7dda3f8806b865139858f8252326/core/src/processing/core/PApplet.java#L1143).

Example GUI console log start:
```
Console Log Started at Local Time: 2021-06-30_16-48-52
Screen Resolution: 1920 X 1080
High-DPI Screen: false
Operating System: Mac
Welcome to the Processing-based OpenBCI GUI!
```

retiutut, 2021-06-30:
> I ran the test using openbci gui master branch and preprocessing 3.5.4
>
> Without any modification, the app is resizable but the resolution is not adapted for uhd screen:
> [image]
>
> When I add the [...]

Counting this post as a successful test! ✅

I have also dug through the Processing code for these methods for completeness.

commit fb33be45 (2021-06-30): Accomodate high-dpi screens Fixes #968
PR #985 (2021-07-31): GUI 5.0.6 - July 2021
# v5.0.6

### Improvements
* Add Auditory Feedback to the Focus Widget Fixes #709

### Bug Fixes
* Fix drawing error in Control Panel WiFi Shield static IP Textfield
* Accomodate high-dpi screens Fixes #968
* Add Arduino Focus Fan example to networking test kit on GitHub repo
* Allow synthetic square wave expert mode keyboard shortcut for Cyton and Ganglion Fixes #976
closed 2021-06-30 by retiutut