OpenBCI/OpenBCI_GUI #976: Some of the GUI Keyboard Shortcuts do not work

Otemo, 2021-06-01:
## Problem
Some of the GUI Keyboard Shortcuts do not work
Explain the problem

## Expected 
under Expert Mode setting...
send '[' ']' to enable/disable synthetic square wave

## Operating System and Version
Windows10

## GUI Version
v5.0.5

## Running standalone app
Both Downloaded app and Running from Processing

## Type of OpenBCI Board
Cyton+Daisy

## Are you using a WiFi Shield?
No

## Console Log
"Expert Mode: '[' pressed. This is not assigned or applicable to current setup"
"Expert Mode: ']' pressed. This is not assigned or applicable to current setup"

retiutut, 2021-06-03:
We can patch this for next month's release. I think you already know @Otemo that you can force these commands using Expert Mode -> Custom Command Textfield -> Send Command.

I think someone, maybe you, mentioned this before and we did not consider Cyton. From Interactivity.pde:

```
    if (currentBoard instanceof BoardGanglion) {
        if (val == '[' ||  val == ']') {
            println("Expert Mode: '" + val + "' pressed. Sending to Ganglion...");
            Boolean success = ((Board)currentBoard).sendCommand(str(val)).getKey();
            if (success) {
                outputSuccess("Expert Mode: Success sending '" + val + "' to Ganglion!");
            } else {
                outputWarn("Expert Mode: Error sending '" + val + "' to Ganglion. Try again with data stream stopped.");
            }
            return;
        }
    }
    ```
    
    
    should be
    
    ```
    if (currentBoard instanceof BoardGanglion || currentBoard instanceof BoardCyton) {
```

Otemo, 2021-06-03:
Yes, it was me.
I'm glad to hear that you will patch it in the next version.
Thank you！

retiutut, 2021-06-30:
Fixed and Tested. ✅

closed 2021-06-30 by retiutut

Otemo, 2022-04-17:
hi retiutut, 
Some commands do not seem to work even with the latest GUI (e.g. v5.0.9). For example, typing 'd' on the keyboard also does not return to the default channel settings (https://docs.openbci.com/Cyton/CytonSDK/). 
Could you please give me some solution to this issues?

retiutut, 2022-04-18:
@Otemo you probably need to turn on “Expert Mode” from the “Settings” Dropdown in the top right of the GUI. This prevents new users from accidentally pressing a key and changing the board. Setting the board to default will also not update the Hardware Settings in the Time Series widget. That’s why such a feature is reserved for “Expert Mode”. This should help. 👍🙂

Otemo, 2022-04-19:
hi retiutut, thank you for all your support.
Yes, I have "Expert Mode" on. However, there seems to be some keys on the keyboard that are not active. One thing I am encountering is that the 'd' key is not working. For example, if I press the '[' key I can see a square wave. But when I press 'd' I cannot see the brain waves. I am using the latest GUI and Cyton board.

retiutut, 2022-04-19:
@Otemo Correct.

```
        case 'd':   
            return;
```

I/we removed this shortcut because of the reason stated above with Hardware Settings being out of sync.

retiutut, 2022-04-19:
We should remove the mention in the docs for this keyboard Shortcut then:

https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/#gui-keyboard-shortcuts

retiutut, 2022-04-19:
I think you might still be able to:
- turn off streaming
- have expert mode on
- open "hardware settings" in Time Series widget
- send custom command as `d` at the bottom of the hardware settings overlay

Otemo, 2022-04-19:
hi retiutut, Thank you for explaining why this change was made and for providing an alternative method. 
I tried the method and it worked!

commit 58a679cf (2021-06-30): Allow enable/disable synthetic square wave expert keyboard shortcut for Cyton and Ganglion

Fixes #976
PR #985 (2021-07-31): GUI 5.0.6 - July 2021
# v5.0.6

### Improvements
* Add Auditory Feedback to the Focus Widget Fixes #709

### Bug Fixes
* Fix drawing error in Control Panel WiFi Shield static IP Textfield
* Accomodate high-dpi screens Fixes #968
* Add Arduino Focus Fan example to networking test kit on GitHub repo
* Allow synthetic square wave expert mode keyboard shortcut for Cyton and Ganglion Fixes #976